syssi/esphome-smg-ii #1: does it work with inverter easun smg II?

amgstone, 2023-02-17:
[SMG-RS232+Communication+Protocol+V1.0(1).pdf](https://github.com/syssi/esphome-pipsolar/files/10740611/SMG-RS232%2BCommunication%2BProtocol%2BV1.0.1.pdf)
 does it work with inverter easun smg II?

syssi, 2023-02-17:
I've prepared two configuration examples:

https://raw.githubusercontent.com/syssi/esphome-smg-ii/main/esp32-example.yaml
https://raw.githubusercontent.com/syssi/esphome-smg-ii/main/esp8266-example.yaml

Please attach the inverter to the `tx_pin` / `rx_pin` and give it a try. The example will request three different registers (operation mode, AC voltage, AC frequency). Please provide some logs! `>>>` indicates outgoind traffic. If the inverter response we should see incoming traffic (`<<<`) at the log.

amgstone, 2023-02-17:
[10:55:38][D][uart_debug:114]: >>> 01:03:02:01:00:03:55:B3
[10:55:38][D][uart_debug:114]: >>> 01:03:02:01:00:03:55:B3
[10:55:38][D][uart_debug:114]: >>> 01:03:02:01:00:03:55:B3
[10:55:38][D][uart_debug:114]: >>> 01:03:02:01:00:03:55:B3
[10:55:38][D][uart_debug:114]: >>> 01:03:02:01:00:03:55:B3
[10:55:39][D][modbus_controller:032]: Modbus command to device=1 register=0x201 countdown=0 no response received - removed from send queue

syssi, 2023-02-17:
I've updated the example configuration. Please flash & try again.

amgstone, 2023-02-17:
[11:11:36][D][uart_debug:114]: >>> 01:03:00:C9:00:03:D5:F5
[11:11:37][D][uart_debug:114]: >>> 01:03:00:C9:00:03:D5:F5
[11:11:37][D][uart_debug:114]: >>> 01:03:00:C9:00:03:D5:F5
[11:11:37][D][uart_debug:114]: >>> 01:03:00:C9:00:03:D5:F5
[11:11:37][D][uart_debug:114]: >>> 01:03:00:C9:00:03:D5:F5
[11:11:37][D][modbus_controller:032]: Modbus command to device=1 register=0xC9 countdown=0 no response received - removed from send queue

syssi, 2023-02-17:
Could you make sure the Modbus RTU address `1` (0x01) is used at the inverter settings and the RS232/ModbusRTU feature is enabled? At the moment the inverter doesn't respond.

amgstone, 2023-02-17:
adress: 001

syssi, 2023-02-17:
Alright. Could you double check your wiring between the ESP and the inverter? Which GPIOs do you use? Did you see the YAML doesn't use GPIO1/GPIO3 (RX/TX) per default? Did you use the correct pinout of the jack? Could you try to swap the RX/TX lines?

amgstone, 2023-02-17:
[image]
[image]
ok
[image]
[image]
[image]
[image]

syssi, 2023-02-17:
Did you change the YAML configuration accordingly? You are using GPIO1 and GPIO3. My example is using GPIO16 and GPIO17 as RX/TX: https://github.com/syssi/esphome-smg-ii/blob/main/esp32-example.yaml#L4-L5

syssi, 2023-02-17:
Do you know the pinout of the RJ45 jack? Is the pinout part of the SMG II manual?

syssi, 2023-02-17:
It looks like the RJ45 is labeled as `RS232` but it's an RS485 port and we should use an RS485 converter module instead of the RS232 module? I'm a bit confused because of the large amount of different informations (different protocols, different interfaces, different device versions, different models).

amgstone, 2023-02-17:
[image]
[image]

amgstone, 2023-02-17:
> 

i conect GPIO16 і GPIO17

syssi, 2023-02-17:
The image is about a 9pin d-sub connector. The inverter has a 8pin RJ45 port.

amgstone, 2023-02-17:
в комплекті є також кабель rg45-db9

amgstone, 2023-02-17:
[image]
original cable

amgstone, 2023-02-17:
[image]

syssi, 2023-02-17:
Could you double check the RJ45 pinout? Your sketch shows pin 1, 2 and 3 is in use. The photo looks like pin 1, 2 and 4 is in use.

amgstone, 2023-02-17:
RJ45 to converor: 1.orange-white, 2.orange, 8. blue-white
convertor tp esp32: yellow-3.3v, purple-gnd, green-rx2, blue-tx2

syssi, 2023-02-17:
Are you sure about pin 8 of your RJ45 jack? It should be **brown** instead of blue-white!

[image]

amgstone, 2023-02-17:
[image]

syssi, 2023-02-17:
Looks good to me!

amgstone, 2023-02-17:
[20:11:09][D][uart_debug:158]: >>> "QPIRI\xC3\xB8T\r"
[20:11:11][D][uart_debug:158]: >>> "QPIGS\xC2\xB7\xC2\xA9\r"
[20:11:13][D][uart_debug:158]: >>> "QMODI\xC3\x81\r"
[20:11:16][D][uart_debug:158]: >>> "QFLAG\xC2\x98t\r"
[20:11:22][D][uart_debug:158]: >>> "QPI\r"
[20:11:24][D][uart_debug:158]: >>> "^P003PI\r"
[20:11:26][D][uart_debug:158]: >>> "^P005PIq\xC2\x8B\r"
[20:11:28][D][uart_debug:158]: >>> "QPI\xC2\xBE\xC2\xAC\r"
[20:11:30][D][uart_debug:158]: >>> "QDIq\e\r"
[20:11:32][D][uart_debug:158]: >>> "QPIRI\xC3\xB8T\r"
[20:11:34][D][uart_debug:158]: >>> "QPIGS\xC2\xB7\xC2\xA9\r"

syssi, 2023-02-17:
It's possible the orange and orange-white cable at the RS233 module must be swapped.

syssi, 2023-02-18:
Could you rephrase your last post? I don't understand you.

syssi, 2023-02-18:
I try to summarize some findings of my research:

1. The RJ45 port of the SMG II is a RS232 port. A MAX3232 is required to convert the logic level from 0/3.3V to -12/+12V.
2. The protocol at the RS232 follows ModbusRTU. The protocol description should apply.
3. The Modbus address is configurable and set to `1` per default
4. The manufacturer sells ESP based WiFi dongles which gets plugged into the RS232 port.
5. The register addresses of the SMG vs. SMX are different.
6. A RS485 converter module shouldn't be used because there is no RS485 port.
7. The register addresses at the protocol description are decimal values not hex.

amgstone, 2023-02-18:
orange and orange-white cable at the RS232 module swapped.

syssi, 2023-02-18:
Could you try to lookup the specs of the MAX3232? I would like to make sure TX and RX are attached to the correct pin of the MAX3232 on both sides (TTL & RS232 levels). In worst-case the wires of both sides must be swapped.

amgstone, 2023-02-18:
I made various rx-tx options, but nothing changes, the inverter does not respond.

syssi, 2023-02-18:
That's a pity. I'm out of ideas.

amgstone, 2023-02-18:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 303, in _keep_alive_loop
    await self._ping()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 628, in _ping
    await self.send_message_await_response(PingRequest(), PingResponse)
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 537, in send_message_await_response
    res = await self.send_message_await_response_complex(
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 522, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.PingRequest'> after 10.0s
INFO Disconnected from ESPHome API for smg-ii.local
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for smg-ii.local: Error resolving IP address: [Errno -5] No address associated with hostname
INFO Trying to reconnect to smg-ii.local in the background
INFO Successfully connected to smg-ii.local
[10:08:59][D][uart_debug:114]: >>> 01:03:00:C9:00:03:D5:F5
[10:08:59][D][modbus_controller.sensor:025]: Sensor new state: 2.00
[10:08:59][D][sensor:127]: 'smg-ii operation mode': Sending state 2.00000  with 0 decimals of accuracy
[10:08:59][D][modbus_controller.sensor:025]: Sensor new state: 2273.00
[10:08:59][D][sensor:127]: 'smg-ii ac voltage': Sending state 227.30000 V with 1 decimals of accuracy
[10:08:59][D][modbus_controller.sensor:025]: Sensor new state: 4993.00
[10:08:59][D][sensor:127]: 'smg-ii ac frequency': Sending state 49.93000 Hz with 2 decimals of accuracy
[10:08:59][D][uart_debug:114]: <<< 01:03:06:00:02:08:E1:13:81:C7:B3
[10:09:09][D][uart_debug:114]: >>> 01:03:00:C9:00:03:D5:F5
[10:09:09][D][modbus_controller.sensor:025]: Sensor new state: 2.00
[10:09:09][D][sensor:127]: 'smg-ii operation mode': Sending state 2.00000  with 0 decimals of accuracy
[10:09:09][D][modbus_controller.sensor:025]: Sensor new state: 2294.00
[10:09:09][D][sensor:127]: 'smg-ii ac voltage': Sending state 229.40001 V with 1 decimals of accuracy
[10:09:09][D][modbus_controller.sensor:025]: Sensor new state: 4994.00
[10:09:09][D][sensor:127]: 'smg-ii ac frequency': Sending state 49.94000 Hz with 2 decimals of accuracy
[10:09:09][D][uart_debug:114]: <<< 01:03:06:00:02:08:F6:13:82:37:B6
[10:09:19][D][uart_debug:114]: >>> 01:03:00:C9:00:03:D5:F5
[10:09:19][D][modbus_controller.sensor:025]: Sensor new state: 2.00
[10:09:19][D][sensor:127]: 'smg-ii operation mode': Sending state 2.00000  with 0 decimals of accuracy
[10:09:19][D][modbus_controller.sensor:025]: Sensor new state: 2278.00
[10:09:19][D][sensor:127]: 'smg-ii ac voltage': Sending state 227.80000 V with 1 decimals of accuracy
[10:09:19][D][modbus_controller.sensor:025]: Sensor new state: 4996.00
[10:09:19][D][sensor:127]: 'smg-ii ac frequency': Sending state 49.96000 Hz with 2 decimals of accuracy
[10:09:19][D][uart_debug:114]: <<< 01:03:06:00:02:08:E6:13:84:B6:71

amgstone, 2023-02-18:
it was my inattention on the converter that confused gnd- +12v.

syssi, 2023-02-18:
I've updated the YAML configuration yesterday. If you flash the updated version there should be a bunch of sensors available.

syssi, 2023-02-18:
Could you provide photos of the final setup (RJ45 + Converter + ESP)? I would like to improve the documentation how to build the data cable.

amgstone, 2023-02-18:
ok

amgstone, 2023-02-18:
Tell me how to add sun power to solar
[image]

SeByDocKy, 2023-02-18:
You did a great job :) .... modbus sensors are done bravo :)

Now time to modbus switches, modbus number, modbus select and modbus text part :)

amgstone, 2023-02-18:
foto
[image]
[image]
[image]
[image]

syssi, 2023-02-18:
Perfect! This helps a lot.

syssi, 2023-02-18:
For the energy dashboard a kWh sensor is required. We have to commulate a power sensor (`W`) to `Wh`. Could you provide another log with all sensors enabled? Could you tell me the power sensor which indicates the amount of power pushed to the mains/grid.

amgstone, 2023-02-18:
i have sensor.smg_ii_pv_charging_average_power I create a sensor Integration - Riemann sum integral and then I add that sensor to solar? Is it possible that I still need to create a utility meter and then add it to solar?

syssi, 2023-02-18:
I don't know which of these sensors is the best / correct one for your goal:

| Description                                            | Unit     |  Res  | Addr | Len | R/W | Values |
| -----------------------------------------------------: | -------: |  ---: | :--: | :-: | :-: | :----- |
| Average mains power                                    |  1W      |  Int  |  204 |  1  |  R  | |
| Average inverter power                                 |  1W      |  Int  |  208 |  1  |  R  | Positive numbers indicate inverter output, negative numbers indicate inverter input |
| Inverter charging power                                |  1W      |  Int  |  209 |  1  |  R  | |
| Output active power                                    |  1W      |  Int  |  213 |  1  |  R  | |
| Battery average power                                  |  1W      |  Int  |  217 |  1  |  R  | |
| PV average power                                       |  1W      |  Int  |  223 |  1  |  R  | |
| PV charging average power                              |  1W      |  Int  |  224 |  1  |  R  | |

You have to attach a `id` to the sensor. For example:

```
  # Inverter charging power                              1W       Int   209 1 R
  - platform: modbus_controller
    modbus_controller_id: smg0
    id: smg0_inverter_charging_power
    name: "${name} inverter charging power"
    address: 209
    register_type: holding
    value_type: S_WORD
    unit_of_measurement: "W"
    accuracy_decimals: 0
```

And feed the sensor measurement f.e. into the `total_daily_energy` component:

```
sensor:
  - platform: total_daily_energy
    name: "${name} energy today"
    restore: true
    icon: mdi:circle-slice-3
    power_id: smg0_inverter_charging_power
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
    unit_of_measurement: kWh

time:
  - platform: sntp
```

amgstone, 2023-02-18:
I can't flash it with a new firmware

```
INFO Reading configuration /config/esphome/pipsolar.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing smg-ii (board: wemos_d1_mini32; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling /data/smg-ii/.pioenvs/smg-ii/src/main.o
Linking /data/smg-ii/.pioenvs/smg-ii/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection14switch_commandERKNS0_20SwitchCommandRequestE+0x8): undefined reference to `esphome::switch_::Switch::turn_on()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection14switch_commandERKNS0_20SwitchCommandRequestE+0xc): undefined reference to `esphome::switch_::Switch::turn_off()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection14select_commandERKNS0_20SelectCommandRequestE+0x0): undefined reference to `esphome::select::SelectCall::set_option(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection14select_commandERKNS0_20SelectCommandRequestE+0x4): undefined reference to `esphome::select::SelectCall::perform()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection16send_switch_infoEPNS_7switch_6SwitchE+0x4): undefined reference to `esphome::switch_::Switch::get_device_class[abi:cxx11]()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection16send_select_infoEPNS_6select6SelectE+0x4): undefined reference to `esphome::select::SelectTraits::get_options[abi:cxx11]() const'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > std::operator+<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&)':
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:6011: undefined reference to `esphome::switch_::Switch::turn_on()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o: in function `esphome::api::APIConnection::switch_command(esphome::api::SwitchCommandRequest const&)':
/config/esphome/.esphome/build/smg-ii/src/esphome/components/api/api_connection.cpp:468: undefined reference to `esphome::switch_::Switch::turn_off()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o: in function `esphome::api::APIConnection::select_command(esphome::api::SelectCommandRequest const&)':
/config/esphome/.esphome/build/smg-ii/src/esphome/components/api/api_connection.cpp:670: undefined reference to `esphome::select::SelectCall::set_option(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/esphome/.esphome/build/smg-ii/src/esphome/components/api/api_connection.cpp:671: undefined reference to `esphome::select::SelectCall::perform()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o: in function `esphome::api::APIConnection::send_switch_info(esphome::switch_::Switch*)':
[... cut ...]
> And feed the sensor measurement f.e. into the `total_daily_energy` component:
> 
> ```
> sensor:
>   - platform: total_daily_energy
>     name: "${name} energy today"
>     restore: true
>     icon: mdi:circle-slice-3
>     power_id: smg0_inverter_charging_power
>     filters:
>       # Multiplication factor from W to kW is 0.001
>       - multiply: 0.001
>     unit_of_measurement: kWh
> 
> time:
>   - platform: sntp
> ```

224 I think it will be good, you need to add that sensor to the firmware

syssi, 2023-02-18:
Done. May be a clean build is required:

```
esphome clean esp32-example.yaml
esphome run esp32-example.yaml
```

amgstone, 2023-02-18:
updated the firmware but it does not compile
```
INFO Reading configuration /config/esphome/pipsolar.yaml...
INFO Detected timezone 'Europe/Kiev'
INFO Generating C++ source...
INFO Compiling app...
Processing smg-ii (board: wemos_d1_mini32; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_frame_helper.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_pb2.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_pb2_service.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_server.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/list_entities.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/proto.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/subscribe_state.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/user_services.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/esp32/gpio.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/esp32/preferences.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/logger/logger.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/md5/md5.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/mdns/mdns_component.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/mdns/mdns_esp32.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/mdns/mdns_esp8266.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/mdns/mdns_rp2040.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/modbus/modbus.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/modbus_controller/modbus_controller.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/modbus_controller/sensor/modbus_sensor.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/network/util.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/ota/ota_backend_arduino_esp32.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/ota/ota_backend_arduino_esp8266.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/ota/ota_backend_arduino_rp2040.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/ota/ota_backend_esp_idf.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/ota/ota_component.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/sensor/automation.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/sensor/filter.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/sensor/sensor.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/socket/bsd_sockets_impl.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/socket/lwip_raw_tcp_impl.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/socket/socket.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_component.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_component_esp32_arduino.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_component_esp8266.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_component_esp_idf.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_component_rp2040.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_debugger.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/wifi/wifi_component.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/wifi/wifi_component_esp32_arduino.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/wifi/wifi_component_esp8266.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/wifi/wifi_component_esp_idf.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/wifi/wifi_component_pico_w.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/application.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/component.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/component_iterator.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/controller.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/helpers.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/log.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/scheduler.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/util.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/main.o
[... 27 lines cut ...]
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/main.o:(.literal._Z5setupv+0xd0): undefined reference to `vtable for esphome::total_daily_energy::TotalDailyEnergy'
[... cut ...]
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:676: undefined reference to `esphome::select::Select::add_on_state_callback(std::function<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int)>&&)'
collect2: error: ld returned 1 exit status
*** [/data/smg-ii/.pioenvs/smg-ii/firmware.elf] Error 1
========================= [FAILED] Took 106.97 seconds =========================
```

syssi, 2023-02-18:
A clean build is required!

amgstone, 2023-02-18:
```
INFO Reading configuration /config/esphome/pipsolar.yaml...
INFO Detected timezone 'Europe/Kiev'
INFO Generating C++ source...
INFO Compiling app...
Processing smg-ii (board: wemos_d1_mini32; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_connection.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_frame_helper.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_pb2.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_pb2_service.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/api_server.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/list_entities.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/proto.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/subscribe_state.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/api/user_services.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/esp32/gpio.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/esp32/preferences.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/logger/logger.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/md5/md5.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/mdns/mdns_component.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/mdns/mdns_esp32.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/mdns/mdns_esp8266.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/mdns/mdns_rp2040.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/modbus/modbus.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/modbus_controller/modbus_controller.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/modbus_controller/sensor/modbus_sensor.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/network/util.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/ota/ota_backend_arduino_esp32.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/ota/ota_backend_arduino_esp8266.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/ota/ota_backend_arduino_rp2040.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/ota/ota_backend_esp_idf.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/ota/ota_component.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/sensor/automation.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/sensor/filter.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/sensor/sensor.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/socket/bsd_sockets_impl.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/socket/lwip_raw_tcp_impl.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/socket/socket.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_component.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_component_esp32_arduino.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_component_esp8266.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_component_esp_idf.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_component_rp2040.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/uart/uart_debugger.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/wifi/wifi_component.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/wifi/wifi_component_esp32_arduino.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/wifi/wifi_component_esp8266.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/wifi/wifi_component_esp_idf.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/components/wifi/wifi_component_pico_w.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/application.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/component.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/component_iterator.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/controller.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/helpers.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/log.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/scheduler.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/esphome/core/util.o
Compiling /data/smg-ii/.pioenvs/smg-ii/src/main.o
Linking /data/smg-ii/.pioenvs/smg-ii/firmware.elf
[... 37 lines cut ...]
/config/esphome/.esphome/build/smg-ii/src/esphome/components/modbus_controller/switch/modbus_switch.h:15: undefined reference to `esphome::switch_::Switch::Switch()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/smg-ii/.pioenvs/smg-ii/src/main.o: in function `esphome::ActionList<esphome::uart::UARTDirection, std::vector<unsigne [... cut ...]
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:676: undefined reference to `esphome::select::Select::add_on_state_callback(std::function<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int)>&&)'
collect2: error: ld returned 1 exit status
*** [/data/smg-ii/.pioenvs/smg-ii/firmware.elf] Error 1
========================= [FAILED] Took 109.59 seconds =========================
```

amgstone, 2023-02-18:
http://joxi.ru/p27E0knhKBQjQr serial number

amgstone, 2023-02-18:
http://joxi.ru/zANBb8nuvDbglA not work

syssi, 2023-02-18:
Could you provide logs which contains the different controls you are trying to use? The raw traffic will help me to debug the issues.

syssi, 2023-02-18:
A screenshot of all entities would be helpful too.

amgstone, 2023-02-18:
[15:22:39][D][modbus_controller.select:014]: New select value 0 from payload
[15:22:39][D][select:015]: 'smg-ii output mode': Sending state Single (index 0)
[15:22:39][D][modbus_controller.select:014]: New select value 3 from payload
[15:22:39][E][modbus_controller.select:034]: No option found for mapping 3
[15:22:39][D][modbus_controller.select:014]: New select value 1 from payload
[15:22:39][D][select:015]: 'smg-ii input voltage range': Sending state Narrow range (index 1)
[15:22:39][D][modbus_controller.select:014]: New select value 0 from payload

amgstone, 2023-02-18:
[15:23:25][D][select:062]: 'smg-ii buzzer mode' - Setting
[15:23:25][D][select:115]: 'smg-ii buzzer mode' - Set selected option to: Beep on faults
[15:23:25][D][modbus_controller.select:048]: Found value 3 for option 'Beep on faults'
[15:23:25][D][uart_debug:114]: >>> 01:01:01:32:00:05:5C:3A
[15:23:25][D][uart_debug:114]: >>> 01:01:01:32:00:05:5C:3A
[15:23:26][D][uart_debug:114]: >>> 01:01:01:32:00:05:5C:3A
[15:23:26][D][modbus_controller:032]: Modbus command to device=1 register=0x132 countdown=0 no response received - removed from send queue
[15:23:26][D][uart_debug:114]: >>> 01:01:01:39:00:01:2C:3B
[15:23:26][D][uart_debug:114]: >>> 01:01:01:39:00:01:2C:3B
[15:23:26][D][uart_debug:114]: >>> 01:01:01:39:00:01:2C:3B
[15:23:26][D][uart_debug:114]: >>> 01:01:01:39:00:01:2C:3B
[15:23:27][D][uart_debug:114]: >>> 01:01:01:39:00:01:2C:3B
[15:23:27][D][modbus_controller:032]: Modbus command to device=1 register=0x139 countdown=0 no response received - removed from send queue
[15:23:27][D][uart_debug:114]: >>> 01:01:01:A4:00:01:BD:D5
[15:23:27][D][uart_debug:114]: >>> 01:01:01:A4:00:01:BD:D5
[15:23:27][D][uart_debug:114]: >>> 01:01:01:A4:00:01:BD:D5
[15:23:27][D][uart_debug:114]: >>> 01:01:01:A4:00:01:BD:D5
[15:23:28][D][uart_debug:114]: >>> 01:01:01:A4:00:01:BD:D5
[15:23:28][D][modbus_controller:032]: Modbus command to device=1 register=0x1A4 countdown=0 no response received - removed from send queue

syssi, 2023-02-18:
Could you try to lookup the serial number of your device using the on screen menu? Could you tell me the serial number of your device?

amgstone, 2023-02-18:
there is no such information in the menu

syssi, 2023-02-18:
I've updated the YAMLs. Please try to use the `switch` and `select` entities again.

amgstone, 2023-02-18:
work good!!!!)))

amgstone, 2023-02-18:
can I put 2 firmwares from bms and inverter on one esp32?

syssi, 2023-02-18:
Yes. Feel free to merge the two YAMLs into one.

syssi, 2023-02-19:
I've pushed some more entities. Could you test the new text sensors decoding the warning and fault codes?

[17 more replies]

closed 2023-03-01 by syssi