syssi/esphome-smg-ii #17: Mention Easun SMX as unsupported

syssi, 2023-02-24:
1. The SMX uses different register addresses. It's compatible to the SRNE probably.
2. The SMX must be interfaced using RS485. There is no RS232 interface.

felipecrs, 2024-10-28:
Hey, @syssi, I have a SRNE HF2430U60-100:

https://www.srnesolar.com/productdetail/Off-Grid-Inverters-HF.html

[image]

Does it mean your integration will not work?

If not, I guess my best bet is what people are doing in https://community.home-assistant.io/t/srne-inverter-integration-help/609730/41?page=2.

syssi, 2024-10-29:
This notice is more about the used Modbus registers. The SMG and SMX are using different Modbus addresses so this project cannot be used without modification. If I remember correctly this use adapted the YAML configuration slightly to the SRNE register addresses: https://github.com/syssi/esphome-smg-ii/issues/28

Do you own the device already?

felipecrs, 2024-10-29:
Yes, I own the device already.

syssi, 2024-10-29:
Just give it a try and feel free to create a new issue to talk about your specific device.

felipecrs, 2024-10-29:
Alright, thanks a ton!

felipecrs, 2024-10-29:
I am waiting for my ESP32, RJ45 to pins adapter, and RS485-TTL to arrive, so it will take me a while before being able to test it.

Out of curiosity, it looks like my inverter is the same as a EASUN ISolar MLV 3KW-U (which also doesn't have a RS232 interface).

Lastly, I have found this thread that mentions the exact same inverter as mine: https://community.home-assistant.io/t/integrating-srne-mppt-inverter-with-ha/490475 and has the Modbus addresses.

syssi, 2024-10-29:
It looks like most of these register addresses does match: https://github.com/syssi/esphome-smg-ii/issues/28#issuecomment-1540066841

The addresses of the SMG-II doesn't match.

felipecrs, 2024-10-29:
This is the pdf that seems to match my inverter btw: https://github.com/shakthisachintha/SRNE-Hybrid-Inverter-Monitor/blob/master/Resources/SRNE%20hybrid%20solar%20inverter%20MODBUS%20protocol%20V1%207.pdf

It's a bit different than the one ([I think](https://github.com/syssi/esphome-smg-ii/issues/28#issuecomment-2444895279)) you used as base for the SRNE example. But there are some matches indeed.