syssi/esphome-smg-ii #19: utility meter gas sensor

amgstone, 2023-03-01:
I'm sorry that this is off topic, I'm trying to configure the gas meter, can you tell me if I configured the utility meter correctly?

esphome:
name: gas
friendly_name: gaz

esp8266:
board: esp01_1m

Enable logging
logger:

Enable Home Assistant API
api:
encryption:
key: "Qak6Up9fGQCVMaqS0IbvN+kkwtiOIMILoji2KSt5+gQ="

ota:
password: "f4acbf42f9ae6b3623f1e6a368f267b2"

wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password

Enable fallback hotspot (captive portal) in case wifi connection fails
ap:
ssid: "Gaz Fallback Hotspot"
password: "SdZrNwwaD5Ns"

time:

platform: sntp
captive_portal:

binary_sensor:

platform: status
name: "gas - Status"
sensor:

platform: wifi_signal
name: "gas - WiFi Signal"
update_interval: 30s

platform: pulse_counter
pin: GPIO2
name: "Gas Cubic meters used"
update_interval : 60s
filters:

multiply: 0.01
unit_of_measurement: "m³/ min"
id: gas
accuracy_decimals: 3
icon: 'mdi:fire'
platform: template
name: "Gas Total m³"
id: Gas_Total_m
lambda: |-
static float total_value = 0.0;
total_value += id(gas).state;
return total_value;
unit_of_measurement: "m³"
icon: 'mdi:fire'
update_interval: 60s
accuracy_decimals: 3

platform: total_daily_energy
name: "gas energy today"
icon: mdi:counter
power_id: Gas_Total_m
accuracy_decimals: 3
unit_of_measurement: kWh
restore: true
filters:

multiply: 0.001

syssi, 2023-03-01:
Could you provide some details about your gas meter? How many pulses does the device send per m³?

amgstone, 2023-03-01:
[image]

amgstone, 2023-03-01:
[image]

syssi, 2023-03-01:
https://www.samgas.com.ua/lichylnyk-gazu-membrannyy-samgaz-g25-rs2001-2
https://www.samgas.com.ua/sites/default/files/_samgaz_re_red_0_6_3_10-12-2018_14.pdf

I've trouble to translate the PDF document. Could you read the document and tell me the page which is about the low frequency pulses?

syssi, 2023-03-01:
> On the last reel after the comma is fixed permanent magnet for working with a low-frequency pulse generator. The 
> design of the counting device has a plastic case in which the counting mechanism itself is assembled with the display
> display drums (Fig. 10):

amgstone, 2023-03-01:
[image]
[image]

amgstone, 2023-03-01:
I install yaml, everything is ok, but as soon as I insert the reader into the gas meter, esp becomes offline.
[image]

amgstone, 2023-03-01:
when the sensor is inserted into the gas meter, esp does not want to install yaml

amgstone, 2023-03-01:
http://joxi.ru/l2ZYZB5Iz9wGlm

amgstone, 2023-03-01:
after unplugging and turning on the power, esp starts counting from the beginning, does not remember the last value

syssi, 2023-03-01:
I will have to build the setup on my own this evening to get an idea what's going on here.

syssi, 2023-03-01:
```
globals:
  - id: total_pulses
    type: int
    restore_value: true
    initial_value: '7470502'

esphome:
  name: gas
  friendly_name: gaz

esp8266:
  board: esp01_1m
  restore_from_flash: true

preferences:
  flash_write_interval: 10min

api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: !secret ap_wifi_ssid
    password: !secret ap_wifi_password

captive_portal:

binary_sensor:
  - platform: gpio
    id: reed
    pin:
      number: GPIO5
    name: "Reed contact closed"
    filters:
      - delayed_on: 20ms
    on_press:
      then:
        - lambda: id(total_pulses) += 1;
        - output.turn_off: led
    on_release:
      then:
        - output.turn_on: led

sensor:
  - platform: template
    name: "Gas Cubic meters used"
    device_class: gas
    unit_of_measurement: "m³"
    state_class: "total_increasing"
    icon: "mdi:fire"
    accuracy_decimals: 2
    lambda: |-
      return id(total_pulses) * 0.001;

output:
  - platform: gpio
    pin: GPIO2
    id: 'led'
```

Please attach the red wire to `GPIO5 / D1` and try again. :-)

syssi, 2023-03-01:
I've tested the setup. It's working and the total pulses are stored every 10 minutes to the flash/rtc:

```
preferences:
  flash_write_interval: 10min
```

Does the LED of the wemos blink once on every round of the gas meter?

amgstone, 2023-03-01:
with d1 does not respond

syssi, 2023-03-01:
Could you try to describe the issue in detail? You could write the summary in your mother language if you like.

amgstone, 2023-03-01:
when it was connected to d4, everything worked, but when I connected the sensor to the esp meter, it did not start. When I connected it to d1, the sensor did not work, did not react to the magnet.

syssi, 2023-03-01:
You have to use a pull-up.

amgstone, 2023-03-02:
and how should it be connected according to the scheme? I have resistors.

syssi, 2023-03-02:
I will paint the schematics later the day!

syssi, 2023-03-02:
I've enabled the internet pull-up of the ESP. Please try this updated YAML:

```
globals:
  - id: total_pulses
    type: int
    restore_value: true
    initial_value: '7470502'

esphome:
  name: gas
  friendly_name: gaz

esp8266:
  board: d1_mini
  restore_from_flash: true

preferences:
  flash_write_interval: 10min

api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: !secret ap_wifi_ssid
    password: !secret ap_wifi_password

captive_portal:

binary_sensor:
  - platform: gpio
    id: reed
    pin:
      number: GPIO5
      mode: INPUT_PULLUP
    name: "Reed contact closed"
    filters:
      - delayed_on: 20ms
    on_press:
      then:
        - lambda: id(total_pulses) += 1;
        - output.turn_off: led
    on_release:
      then:
        - output.turn_on: led

sensor:
  - platform: template
    name: "Gas Cubic meters used"
    device_class: gas
    unit_of_measurement: "m³"
    state_class: "total_increasing"
    icon: "mdi:fire"
    accuracy_decimals: 2
    lambda: |-
      return id(total_pulses) * 0.001;

output:
  - platform: gpio
    pin: GPIO2
    id: 'led'
```

amgstone, 2023-03-02:
work perfectly

amgstone, 2023-03-02:
http://joxi.ru/gmvVao0sqYL5bm when I change the value to the current one and flash esp, nothing changes in HA, in the sensor:
sensor.gas_gas_cubic_meters_used

syssi, 2023-03-02:
I assume this multiplicator is wrong:

```
    lambda: |-
      return id(total_pulses) * 0.001;
```

Could you check how many times the reed contact is closed while the last digit of the gas meter is spinning? I assume it's closed once per round. So the multiplicator should be:

```
    lambda: |-
      return id(total_pulses) * 0.01;
```

amgstone, 2023-03-02:
http://joxi.ru/Q2KqbERuLD4NKr then when i changed the value to 0.01

amgstone, 2023-03-02:
initial values ​​can be changed in HA utility meter. But you need to add it to yaml.

syssi, 2023-03-02:
Did you see the YAML above how to set the initial value of `total_pulses` using a temporary added `button` entity?

amgstone, 2023-03-02:
http://joxi.ru/82Q4bdkuwejP0m work))))

syssi, 2023-03-02:
Cool! Please compare the value of the gas meter with the value of the ESPHome node in a few days to verify the multiplication is correct and the ESPHome increments the counter just as fast as the gas meter.

closed 2023-03-02 by syssi

amgstone, 2023-03-02:
and the utility meter does not need to be added for statistics?

syssi, 2023-03-02:
No. It's useless in my opinion and shouldn't be added to the YAML.

syssi, 2023-03-02:
The `Gas Cubic meters used` sensor should be sufficient.

amgstone, 2023-03-11:
for some reason, some impulses are present when there is no gas consumption
[image]

syssi, 2023-03-11:
How long is the cable between the ESP and the reed contact?

amgstone, 2023-03-11:
2 meter

syssi, 2023-03-11:
The enabled internal resistor isn't probably enough and you could increase the debounce time.

amgstone, 2023-03-11:
- delayed_on: 20ms, how much to change the value?

syssi, 2023-03-11:
Please try 150ms. If you miss pulses decrease the value step by step.