syssi/esphome-smg-ii #28: SRNE ASF Support

Fidcom, 2023-05-09:
Hi, this may be off topic, but out of all your inverter integrations, this one is the most similar for the srne inverter, and some settings works better compared to the pdf protocol doc. I was able to get all necessary entities to work, but the ones related to DC voltage such as the float, bulk settings, etc., I can only read them, but not write. 

```
number:
  - platform: modbus_controller
    modbus_controller_id: snre
    name: Bulk Charge Voltage
    id: bulk_voltage
    address: 0xE008
    value_type: U_WORD
    register_type: holding
    unit_of_measurement: "V"
    min_value: 48
    max_value: 58.4
    step: 0.4
    lambda: "return  x * 0.4; "      
    write_lambda: "return  x * 2.5; "   
    use_write_multiple: true

  - platform: modbus_controller
    modbus_controller_id: snre
    name: Float Charge Voltage 
    id: float_voltage
    address: 0xE009
    value_type: U_WORD
    register_type: holding
    unit_of_measurement: "V"
    min_value: 48
    max_value: 58.4
    step: 0.4
    lambda: "return  x * 0.4; "      
    write_lambda: "return  x * 2.5; "
``` 
When I try to adjust the slider, the error I receive is the following, and the next one if I dont use the "use_write_multiple" function.

```
Modbus error function code: 0x90 exception: 11 
[21:46:53][E][modbus_controller:068]: Modbus error function code: 0x10 exception: 11 
[21:46:53][E][modbus_controller:077]: Modbus error - last command: function code=0x10 register address = 0xE008 registers count=1 payload size=2

[21:55:48][D][modbus:119]: Modbus error function code: 0x86 exception: 11
[21:55:48][E][modbus_controller:068]: Modbus error function code: 0x6 exception: 11 
[21:55:48][E][modbus_controller:077]: Modbus error - last command: function code=0x6  register address = 0xE009  registers count=1 payload size=2
```
I don't have this problem with other settings to adjust charge current, etc. Therefore, do you have any idea what the problem might be based on the modbus error? Thank you!

Fidcom, 2023-05-09:
[image]

This is the reason, the values needs to be converted to 12V

Fidcom, 2023-05-09:
I'm still testing some entities to see which ones are useful for me but this is the configuration I'm using together with the jk bms

```
substitutions:
  name: inverter
  device_description: "Monitor and control a JK-BMS via UART-TTL"
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO03
  rx_pin: GPIO01
  mac_address: C8:47:8C:E9:2C:58
  # Defaults to "JK02"
  # Please use "JK02_32S" if you own a JK-B2A8S20P >= hardware version 11+ (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.4.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf

external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s

logger:
  baud_rate: 0
#  level: DEBUG

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("JK-", 0) == 0) {
            ESP_LOGI("ble_adv", "New JK-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 5s
    id: bms0

uart:
  - id: mod_bus
    tx_pin: 16
[... 972 lines cut ...]
        case 5: return std::string("Charging");
        case 6: return std::string("Fault");
      }
      return std::string("Unknown");

  - platform: modbus_controller
    modbus_controller_id: snre
    name: "Charge State"
    id: charge_state
    address: 0x010B
    register_type: holding
    raw_encode: HEXBYTES
    lambda: |-
      uint16_t value = modbus_controller::word_from_hex_str(x, 0);
      switch (value) {
        case 0: return std::string("Charge Off");
        case 1: return std::string("Quick Charge");
        case 2: return std::string("Cons Volt Charge");
        case 4: return std::string("Float Charge");
        case 5: return std::string("Reserved");
        case 6: return std::string("Li Batt Activate");
        case 7: return std::string("Reserved");
      }
      return std::string("Unknown"); 
```

syssi, 2023-06-25:
I would like to close this issue some day. Could you provide a summary including an updated YAML? What's working and what's not working yet? Thanks in advance!

Fidcom, 2023-06-25:
The yaml is the same, I cant solve the problem because i dont know how to solve the modbus errors. The errors only shows when I try to change the dc battery values like float, bulk, cutoff voltages, etc. For the other settings like ac current, charge source, etc, it works ok.

```
Modbus error function code: 0x90 exception: 11 
[21:46:53][E][modbus_controller:068]: Modbus error function code: 0x10 exception: 11 
[21:46:53][E][modbus_controller:077]: Modbus error - last command: function code=0x10 register address = 0xE008 registers count=1 payload size=2

[21:55:48][D][modbus:119]: Modbus error function code: 0x86 exception: 11
[21:55:48][E][modbus_controller:068]: Modbus error function code: 0x6 exception: 11 
[21:55:48][E][modbus_controller:077]: Modbus error - last command: function code=0x6  register address = 0xE009  registers count=1 payload size=2
```

syssi, 2023-06-25:
Do you know any other (may be official software) which is able to control these values? We could try to sniff/capture some frames to compare the payload with the ESPHome modbus implementation. Do you own a OEM WiFi dongle for example?

Fidcom, 2023-06-25:
Currently I do not have the wifi module and I also installed the inverter to a friend, I can only access it remotely through home assistant. I wrote to SRNE several times but never received a response. Searching for SRNE modbus protocol documents I did not find any that were more recent but some older ones. Although the description is more for the charge controller(but  the address are the same), in this document is the following:

[image]

**The inverter manual shows:**

[image]

I tested on User and the others lithium settings but I dont remeber if i tested with SLD settings. I will try later with other battery types settings to know if that solve the modbus error, but I need to be there in case of battery settings change that i cant revert remotely.

felipecrs, 2024-10-30:
[Another thread](https://github.com/HotNoob/PythonProtocolGateway/issues/35#issuecomment-2445114766) says that:

[SRNE Solar Charge Inverter MODBUS Protocol1.96.pdf](https://github.com/user-attachments/files/17561920/SRNE.Solar.Charge.Inverter.MODBUS.Protocol1.96.pdf)

Is the right manual for ASP/ASF series.

But I contacted SRNE support, and they also confirmed that such manual also applies to my inverter SRNE: HF2430U60-100 (HF series).

alanv72, 2024-11-17:
> I'm still testing some entities to see which ones are useful for me but this is the configuration I'm using together with the jk bms
>
> ```
> substitutions:
>   name: inverter
>   device_description: [...]

Can you share your dashboard for this?

closed 2024-12-26 by syssi