syssi/esphome-smg-ii #33: Support for Cosuper SPE5000?

bartixmayo, 2023-07-21:
Great job! I wonder if this configuration would work with Cosuper SPE5000-PR248L inverter with the Modbus protocol mentioned below.
[Cosuper Modbus protocol.pdf](https://github.com/syssi/esphome-smg-ii/files/12117739/Cosuper.Modbus.protocol.pdf)
[image]

syssi, 2023-07-21:
The protocol descriptions looks good. The device uses Modbus communication and adjusting the registers addresses should be sufficient to talk to the device. Do you own the inverter and do you want to give it a try?

bartixmayo, 2023-07-21:
I am very interested to buy this inverter. If in your opinion there is a
big chance of successful integration I would buy it, check and confirm.