syssi/esphome-smg-ii #37: how can you adjust the current for charging from the network?

amgstone, 2023-09-25:
how can you adjust the current for charging from the network? there is no such setting

amgstone, 2023-09-26:
[image]
read and write 
reading, it is still necessary to record, change

syssi, 2023-10-25:
@heckmic Please subscribe this issue. It's about making 

```
Maximum mains charging current                        0.1A    Uint  333 1 R/W
```

writable.

amgstone, 2023-10-25:
when can you add? a very useful function

syssi, 2023-10-26:
My notes to simulate and test the feature locally.

```
[19:53:13][D][uart_debug:114]: >>> 01:03:01:4B:00:07:75:E2
# 01:4B -> 331
# 00:07 ->   7

[19:53:13][D][modbus_controller.select:014]: New select value 1 from payload
[19:53:13][D][select:015]: 'pow-hvm battery charging priority': Sending state PV priority (index 1)
[19:53:13][D][modbus_controller.sensor:025]: Sensor new state: 400.00
[19:53:13][D][sensor:127]: 'pow-hvm maximum charging current': Sending state 40.00000 A with 1 decimals of accuracy
[19:53:13][D][modbus_controller.sensor:025]: Sensor new state: 400.00
[19:53:13][D][sensor:127]: 'pow-hvm maximum mains charging current': Sending state 40.00000 A with 1 decimals of accuracy
[19:53:13][D][modbus_controller.sensor:025]: Sensor new state: 584.00
[19:53:13][D][sensor:127]: 'pow-hvm Eq Charging voltage': Sending state 58.40000 V with 1 decimals of accuracy
[19:53:13][D][modbus.number:023]: Number new state : 60.00
[19:53:13][D][number:012]: 'pow-hvm battery equalization time': Sending state 60.000000
[19:53:13][D][modbus.number:023]: Number new state : 120.00
[19:53:13][D][number:012]: 'pow-hvm equalization Timeout exit': Sending state 120.000000
[19:53:13][D][modbus.number:023]: Number new state : 30.00
[19:53:13][D][number:012]: 'pow-hvm two equalization charging intervals': Sending state 30.000000
[19:53:13][D][uart_debug:114]: <<< 01:03:0E:00:01:01:90:01:90:02:48:00:3C:00:78:00:1E:09:5F

0x01 0x03
0x0E        # 14 bytes
0x00 0x01   # 331 (1)       charging priority
0x01 0x90   # 332 (400)     maximum charging current
0x01 0x90   # 333 (400)     maximum mains charging current
0x02 0x48   # 334 (584)     eq charging voltage
0x00 0x3C   # 335 (60)      battery equalization time
0x00 0x78   # 336 (120)     equalization timeout exit
0x00 0x1E   # 337 (30)      two equalization charging intervals
0x09 0x5F   # CRC
```

syssi, 2023-10-26:
Please add these two number entities to your YAML and give it a try!

https://github.com/syssi/esphome-smg-ii/pull/42/files

amgstone, 2023-10-27:
work good, thenks)))

PR #42 (2023-10-27): Make maximum charging and maximum mains charging current controllable
Fixes: #37
closed 2023-10-27 by syssi