syssi/esphome-smg-ii #41: Modbus device=1 set offline

studiowena, 2023-10-26:
Hi,

I have strange behavor in communication esp32 - HA esphome > SMG II modbus
In logs i have

`[16:23:26][W][modbus_controller:030]: Modbus device=1 set offline
[16:23:26][D][modbus_controller:043]: Modbus command to device=1 register=0x131 countdown=0 no response received - removed from send queue
[16:23:26][D][uart_debug:114]: >>> 01:03:01:39:00:01:55:FB
[16:23:27][D][uart_debug:114]: >>> 01:03:01:39:00:01:55:FB
[16:23:27][D][uart_debug:114]: >>> 01:03:01:39:00:01:55:FB
[16:23:27][D][uart_debug:114]: >>> 01:03:01:39:00:01:55:FB
[16:23:27][D][uart_debug:114]: >>> 01:03:01:39:00:01:55:FB
[16:23:27][D][modbus_controller:043]: Modbus command to device=1 register=0x139 countdown=0 no response received - removed from send queue`

and then goes online

`[16:23:39][D][modbus_controller:043]: Modbus command to device=1 register=0x149 countdown=0 no response received - removed from send queue
[16:23:39][D][uart_debug:114]: >>> 01:03:01:4B:00:07:75:E2
[16:23:40][D][uart_debug:114]: >>> 01:03:01:4B:00:07:75:E2
[16:23:40][D][uart_debug:114]: >>> 01:03:01:4B:00:07:75:E2
[16:23:40][W][modbus_controller:064]: Modbus device=1 back online`

when its online sensor are read and refresh on home assistant but when goes offline not... and have data with delay.

is this problem with esphome firmware? dont have this problem before...

syssi, 2023-10-26:
Could you provide your YAML and the used ESPHome version? I'm a bit confused: Why is there just outgoing traffic printed to the log and there are no log messages about incoming traffic?

studiowena, 2023-10-26:
Hi,

It was working good from start... but now these warnings and data is with delay cuz of this offline / online changes

INFO ESPHome 2023.10.3

my SMG II Yaml
[https://we.tl/t-IWnPPpiKUE](url)

and log from start
[https://we.tl/t-u02Y3VKF5U](url)

syssi, 2023-10-26:
This is a recently added (new) core feature. See https://github.com/esphome/esphome/pull/5318

Could you try something like:

```
modbus_controller:
  - id: smg0
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s
    offline_skip_updates: 50
```

studiowena, 2023-10-26:
i've changed this section to Yours, 

on ESPHome - clean build files, validate and install, compiled and it run but stops comunicate with modbus... esp is online but not comunicate at all

logs
[https://we.tl/t-FQ6LJohHSH](url)

strange, dont know

studiowena, 2023-10-26:
just hashed this option

`offline_skip_updates: 50`

and it run again, slow with this

```
Modbus device=1 set offline
Modbus command to device=1 register=0x131 countdown=0 no response received - removed from send queue
```

but at least update sensors, but it slow... so still dont know where is the problem

syssi, 2023-10-26:
I will have another look tomorrow again. In worst case we could go back to a `modbus_controller` pre 2023.9.0 without this new feature.

@0x3333 It is possible to disable the new feature?

0x3333, 2023-10-26:
This concept was added to ESPHome, where if a module doesn't respond to a request, it is marked as offline, and if `offline_skip_updates` is set, it will skip this number of updates. If you don't want this feature, set `offline_skip_updates` to zero. It will keep saying it is offline but will keep sending requests.

syssi, 2023-10-26:
Thanks for your quick response. So the "device is marked as offline" feature isn't new?

0x3333, 2023-10-26:
> Thanks for your quick response. So the "device is marked as offline" feature isn't new?

It was added last month.

0x3333, 2023-10-26:
But as I said, if you set `offline_skip_updates` to zero(Which is by default) the behavior is the same as before.

0x3333, 2023-10-26:
I believe your problem is related to `command_throttle`, not `offline_skip_updates`. Remove `command_throttle` and `offline_skip_updates`, it will keep sending requests.

[image]

studiowena, 2023-10-26:
ok ill try and will post info here

studiowena, 2023-10-26:
just installed without this option command_throttle but still has in logs

```
[21:29:22][W][modbus_controller:030]: Modbus device=1 set offline
[21:29:22][D][modbus_controller:043]: Modbus command to device=1 register=0x6C countdown=0 no response received - removed from send queue
[21:29:22][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[21:29:23][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[21:29:23][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[21:29:23][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[21:29:23][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[21:29:23][D][modbus_controller:043]: Modbus command to device=1 register=0xC9 countdown=0 no response received - removed from send queue
```

Watt sensor is updating sometimes in 4-5 seconds but next time waiting 30-50 seconds... its not normall behavor, dont know whats wrong ... logs tells that its some info skipped from queue...

0x3333, 2023-10-26:
Well, it looks like the device is not responding... not an issue with ESPHome itself. See the uart_debug, it is sending the read command. I don't know about your device.

Did you tried reverting ESPHome to 2023.09?

studiowena, 2023-10-26:
Im reverting to INFO ESPHome 2023.8.3 - this backup i have. I'll let You know result for fe minutes

my device is

ESP32 ESP-WROOM-32U
WiFi 4dBi do ESP32 U.FL
RS232 TTL MAX3232 DB9

its connected to SMG II inverter

studiowena, 2023-10-26:
reverted to 2023.8.3

yaml

```
modbus_controller:
  - id: smg0
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s
```

logs

```
INFO ESPHome 2023.8.3
INFO Reading configuration /config/esphome/smg-ii.yaml...
INFO Detected timezone 'Europe/Warsaw'
INFO Starting log output from smg-ii.local using esphome API
INFO Successfully connected to smg-ii.local
[22:17:35][I][app:102]: ESPHome version 2023.8.3 compiled on Oct 26 2023, 22:05:29
[22:17:35][I][app:104]: Project syssi.esphome-smg-ii version 1.0.0
```

still on logs is

```
[22:17:38][D][modbus_controller:032]: Modbus command to device=1 register=0xC9 countdown=0 no response received - removed from send queue
```
but no offline / online changes, 

so this is problem on communicate from modbus to esp32 or from esp32 to modbus? what direction?

0x3333, 2023-10-26:
Well, looks like you have a comm problem, not an ESPHome one. I can't help you with that, sorry. Someone with knowledge in this project may help.

studiowena, 2023-10-27:
Hi again,

After reseting SMG II Inverter, no warnings on ESPHome logs and data from modbus is readed in 10s intervals so it works OK now... 

Dont know SMG II that much to know if this problem with modbus data is common and one restarting solve problem forever... 

I will monitor and post here if problem back.

closed 2023-11-04 by syssi