syssi/esphome-smg-ii #43: Add additional controls of writable registers

syssi, 2023-10-27:
```
  # Output voltage                                        0.1V    Uint  320 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} output voltage"
    address: 320
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    # state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

  # Output frequency setting                              0.01Hz  Uint  321 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} output frequency setting"
    address: 321
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "Hz"
    device_class: frequency
    # state_class: measurement
    accuracy_decimals: 2
    filters:
      - multiply: 0.01

  # Battery overvoltage protection point                  0.1V    Uint  323 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} battery overvoltage protection point"
    address: 323
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    # state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

  # Max charging voltage                                  0.1V    Uint  324 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} max charging voltage"
    address: 324
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    # state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

  # Floating charging voltage                             0.1V    Uint  325 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} floating charging voltage"
    address: 325
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    # state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

  # Battery discharge recovery point in mains mode        0.1V    Uint  326 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} battery discharge recovery point in mains mode"
    address: 326
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    # state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

  # Battery low voltage protection point in mains mode    0.1V    Uint  327 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} battery low voltage protection point in mains mode"
    address: 327
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    # state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

  # Battery low voltage protection point in off-grid mode 0.1V    Uint  329 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} battery low voltage protection point in off-grid mode"
    address: 329
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    # state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

  # Maximum charging current                              0.1A    Uint  332 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} maximum charging current"
    address: 332
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "A"
    device_class: current
    # state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

  # Maximum mains charging current                        0.1A    Uint  333 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} maximum mains charging current"
    address: 333
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "A"
    device_class: current
    # state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

  # Eq Charging voltage                                   0.1V    Uint  334 1 R/W
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} Eq Charging voltage"
    address: 334
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    # state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1
```

studiowena, 2023-10-27:
is there any chance to get info from modbus of uptime time?

amgstone, 2023-10-27:
Good afternoon, I decided to switch to wt32-eth01, I use 2 inverters, tell me which pins to connect to using wt32-eth01. now I have esp32, I use pins: 
tx_pin: GPIO16
rx_pin: GPIO17
tx_pin2: GPIO32
rx_pin2: GPIO33

amgstone, 2023-10-29:
burned 4 esp32 boards, for some reason they boiled after the update

syssi, 2023-10-29:
Did you change the wiring? A software update cannot destroy the ESP.

amgstone, 2023-10-29:
i didn't change wiring.

syssi, 2023-10-29:
Did you change something else?

amgstone, 2023-10-29:
no(((

syssi, 2023-10-29:
What happens if you revert/remove the lines above again?

amgstone, 2023-10-29:
I don't have an esp32 at the moment, I'll pick it up on Monday and experiment

syssi, 2023-11-04:
@studiowena the inverter doesn't expose any details about the uptime unfortunately.

amgstone, 2023-11-04:
replaced all the ingredients and everything works

syssi, 2023-11-04:
Did you try the new controls another time or are you afraid? :-)

amgstone, 2023-11-04:
yes, I wrote the automation for power reshulers

amgstone, 2023-11-04:
alias: Лімітер зарядки акума від мережі
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.smg_output_power
condition:
  - condition: state
    entity_id: sensor.smg_ii_operation_mode_id
    state: "2"
action:
  - choose:
      - conditions:
          - condition: numeric_state
            entity_id: sensor.smg_output_power
            below: 500
        sequence:
          - service: number.set_value
            data:
              value: "40"
            target:
              entity_id: number.smg_smg_jk_maximum_mains_charging_current
      - conditions:
          - condition: numeric_state
            entity_id: sensor.smg_output_power
            below: 1000
            above: 499.9
        sequence:
          - service: number.set_value
            data:
              value: "35"
            target:
              entity_id: number.smg_smg_jk_maximum_mains_charging_current
      - conditions:
          - condition: numeric_state
            entity_id: sensor.smg_output_power
            below: 1500
            above: 1000
        sequence:
          - service: number.set_value
            data:
              value: "30"
            target:
              entity_id: number.smg_smg_jk_maximum_mains_charging_current
      - conditions:
          - condition: numeric_state
            entity_id: sensor.smg_output_power
            below: 2000
            above: 1500
        sequence:
          - service: number.set_value
            data:
              value: "25"
            target:
              entity_id: number.smg_smg_jk_maximum_mains_charging_current
      - conditions:
          - condition: numeric_state
            entity_id: sensor.smg_output_power
            below: 2500
            above: 2000
        sequence:
          - service: number.set_value
            data:
              value: "20"
            target:
              entity_id: number.smg_smg_jk_maximum_mains_charging_current
      - conditions:
          - condition: numeric_state
            entity_id: sensor.smg_output_power
            below: 3000
            above: 2500
        sequence:
          - service: number.set_value
            data:
              value: "15"
            target:
              entity_id: number.smg_smg_jk_maximum_mains_charging_current
      - conditions:
          - condition: numeric_state
            entity_id: sensor.smg_output_power
            below: 3500
            above: 3000
        sequence:
          - service: number.set_value
            data:
              value: "10"
            target:
              entity_id: number.smg_smg_jk_maximum_mains_charging_current
      - conditions:
          - condition: numeric_state
            entity_id: sensor.smg_output_power
            below: 4000
            above: 3500
        sequence:
          - service: number.set_value
            data:
              value: "5"
            target:
              entity_id: number.smg_smg_jk_maximum_mains_charging_current
      - conditions:
          - condition: numeric_state
            entity_id: sensor.smg_output_power
            below: 4500
            above: 4000
        sequence:
          - service: number.set_value
            data:
              value: "5"
            target:
              entity_id: number.smg_smg_jk_maximum_mains_charging_current
      - conditions:
          - condition: numeric_state
            entity_id: sensor.smg_output_power
            above: 5000
        sequence:
          - service: number.set_value
            data:
              value: "2"
            target:
              entity_id: number.smg_smg_jk_maximum_mains_charging_current
mode: single

PR #44 (2023-11-04): Add additional controls of writable registers
Fixes: #43
closed 2023-11-04 by syssi