syssi/esphome-smg-ii #47: not work 2 esp32

amgstone, 2023-10-30:
I installed the patch on the second esp32 but it does not work, tell me what I did wrong. uart,modbus replaced, but gives an error
http://joxi.ru/L21JNyKSREpYdA

substitutions:
  name: smg-ii-2
  device_description: "Monitor and control a ISolar/EASUN SMG II inverter via RS232"
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-smg-ii-2"
    version: 1.0.0

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: "Irena_house_router"
  password: "1212121212"

  manual_ip:
    static_ip: 192.168.31.150
    gateway: 192.168.31.1
    subnet: 255.255.255.0

logger:
  level: INFO
  logs:
    component: ERROR

api:
  encryption:
    key: "MHfWTYawN5KNNWbxeDRTen6FZYhD+YMMZzKyhUE9f40="

ota:
  password: "f0700bdc485f12a2f8bf7e7ef660f871"

uart:
  - id: uart_1
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

modbus:
  - id: modbus1
    uart_id: uart_1
    send_wait_time: 200ms

modbus_controller:
  - id: smg1
    address: 0x01
    modbus_id: modbus1
    command_throttle: 200ms
    update_interval: 10s

time:
  - platform: sntp
    id: my_time

sensor:
  - platform: total_daily_energy
    name: "${name} PV energy today"
    restore: true
    icon: mdi:counter
    power_id: smg1_pv_average_power
[... 886 lines cut ...]
    use_write_multiple: true
    address: 336
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 900.0
    step: 1
    unit_of_measurement: "min"

  # Two equalization charging intervals                           day     Uint  337 1 R/W Range: 1~90
  - platform: modbus_controller
    modbus_controller_id: smg1
    name: "${name} two equalization charging intervals"
    use_write_multiple: true
    address: 337
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 90.0
    step: 1
    unit_of_measurement: "day"

# Button entities

  # Exit t

syssi, 2023-11-05:
Could you explain in detail what's your goal? Do you want to Interface two inverters using one ESP and s single Modbus / RS485 converter?

syssi, 2023-11-05:
May be I got the point and the issue is resolved already? Your setup is up and running again and you are able to use the new features / controls.

closed 2023-11-05 by syssi

amgstone, 2023-11-05:
good afternoon, now everything is resolved, I replaced two pieces of esp32 and a ttl adapter. everything is working now. I have one esp32 and ttl adapter connected to each inverter.

amgstone, 2023-11-05:
I have: WT32-ETH01. I want to squeeze two inverters in there