syssi/esphome-smg-ii #49: Frequency display 500 HZ and twice int 234

S2000Stefan, 2023-11-26:
Hello,
I installed the update 1.1.0 on my SMG II today.  I noticed that in the esp32-example.yaml the section Int 234 is included twice. Is this intentional? 
And that the adjustable frequency display is set to 500.0 HZ in Home Assistant. But it should be set to 50 HZ. :)
Perhaps this is because my inverter is only acting as a charge controller at the moment and is in standby mode?

The update brings interesting new features. 👍  :) 

[image]
[image]

syssi, 2023-11-26:
Thanks for your feedback!

> I noticed that in the esp32-example.yaml the section Int 234 is included twice. Is this intentional?

It's not. I'v removed the duplicate.

syssi, 2023-11-26:
> And that the adjustable frequency display is set to 500.0 HZ in Home Assistant. But it should be set to 50 HZ. :)

Good catch! I've pushed another fix to shift the decimal place by one.

syssi, 2023-11-26:
> Perhaps this is because my inverter is only acting as a charge controller at the moment and is in standby mode?

I don't know. These are the known operation modes:

https://github.com/syssi/esphome-smg-ii/blob/main/esp32-example.yaml#L898-L916

Is the operation mode at the on screen display visible too?

S2000Stefan, 2023-11-26:
Das nenne ich mal einen support, kaum online gestellt und schon gefixt. Vielen dank. 💯 

Das mit dem Standby mode war nur eine Vermutung von mir das deshalb die Frequenzanzeige nicht richig ist. 
Dem ist wohl nicht so.  :) 
Auf dem Display wird das nicht angezeigt.

closed 2023-11-26 by S2000Stefan