syssi/esphome-smg-ii #53: Easun SMH-II support?

hojnikb, 2024-01-19:
Would it be possible to add Easun SMH-II support? Looks like a smiliar unit to SMG-II and uses the common SmartESS app and wifi-pro module.

Thank you!

syssi, 2024-01-19:
Do you own the device and could you give it a try using this project?

hojnikb, 2024-01-19:
> Do you own the device and could you give it a try using this project?

Yes, i own the the device. I will give it a try and report back.

hojnikb, 2024-01-23:
> Do you own the device and could you give it a try using this project?

One question; I've hooked up my rs232-ttl converter to a usb serial. If everything worked, would it be expected for something to come out of the terminal like putty? I've set the baudrate to 2400, as in the script, but i'm not getting anything.

hojnikb, 2024-01-23:
> No, you have to request data to receive a response (on both cases: RS232 vs RS485)!

Oh. that makes sense. Is there some way to identify what sort of communication layer is used (232 or 485)? 

I'm guessing the provided wifi plug pro-05 can support both ?

hojnikb, 2024-03-24:
So i did some playing around a bit more. After i've dissasablmed the wifi plug-pro05 adapter, i've figured that this is almost certanly RS232 interface.

After playing around with this script a bit, couldn't make it to work.

But i did something else. I've taken the wifiplug, used a rs232-ttl adapter and a usb-serial converter, soldered the RX/TX wires to the wifi plug directly and fired up hercules.

After that, i'm certanly getting some information, but can't make any sense of it to be fair. But it does confirm, this is indeed RS232.

Example of output i get:

Data it transmits on its own after some interval:

`(PI30š(553555355535552ý(1455355535553555355535(VERFW:00020.01.(VERFW2:00000.00¶Ű(230.0 13.9 230.0 50.0 13.9 3200 3000 24.0 23.0 21.0 29.2 27.0 1 30 030 0 2 1 6 01 0 0 27.0 0 1šX(EakyDbjuvxzÔŻ(238.1 49.9 230.1 49.9 0006 0006 000 358 23.80 001 054 0045 0000 000.0 00.00 00000 00010000 00 00 00000 010V(BçÉ(00000000000000000000000000000000ëä(230.0 50.0 0030 21.0 27.0 28.2 23.0 50 0 0 2 0 0 0 0 0 1 1 0 0 1 0 27.0 0 1ëś(010 020 030 040 050 060 070 080 090 100 110¸"(002 010 020 030 040 050 060 070 080vb(00ˇ`

And output i get, if i request "Output source priority rating"

`(230.0 13.9 230.0 50.0 13.9 3200 3000 24.0 23.0 21.0 29.2 27.0 1 30 030 0 2 1 6 01 0 0 27.0 0 1šX(230.0 13.9 230.0 50.0 13.9 3200 3000 24.0 23.0 21.0 29.2 27.0 1 30 030 0 2 1 6 01 0 0 27.0 0 1šX`

I could try getting you HEX formats or some other terminal app, if that would be of any help.

hojnikb, 2024-03-24:
I'm guessing these are the commands wifiplug sends (i've reversed tx/rx to get that)
`{51}{50}{49}{BE}{AC}{0D}{51}{49}{44}{D6}{EA}{0D}{51}{53}{49}{44}{BB}{05}{0D}{51}{56}{46}{57}{62}{99}{0D}{51}{56}{46}{57}{32}{C3}{F5}{0D}{51}{50}{49}{52}{49}{F8}{54}{0D}{51}{46}{4C}{41}{47}{98}{74}{0D}{51}{50}{49}{47}{53}{B7}{A9}{0D}{51}{4D}{4F}{44}{49}{C1}{0D}{51}{50}{49}{57}{53}{B4}{DA}{0D}{51}{44}{49}{71}{1B}{0D}{51}{4D}{43}{48}{47}{43}{52}{D8}{55}{0D}{51}{4D}{55}{43}{48}{47}{43}{52}{26}{34}{0D}{51}{4F}{50}{4D}{A5}{C5}{0D}`

syssi, 2024-03-24:
The protocol is called PI30 and probably supported by the pipsolar component. Please give it a try!

hojnikb, 2024-03-24:
> The protocol is called PI30 and probably supported by the pipsolar component. Please give it a try!

Thank you! It appears to be working.

closed 2024-03-25 by syssi

MishuPtm, 2024-03-26:
Hey @hojnikb would you mind sharing the config that worked for you? I have the same inverter and would like to get it working.

hojnikb, 2024-03-27:
> Hey @hojnikb would you mind sharing the config that worked for you? I have the same inverter and would like to get it working.

Hi,

example config work righ out of the box for me
https://github.com/syssi/esphome-pipsolar/blob/main/esp32-example.yaml