syssi/esphome-smg-ii #54: Ardunio Code for esp32

jayprojects, 2024-02-03:
Hi,
Just wanted to thank you for this awesome project. I tested your esp32 example yaml on EspHome. Worked like magic! No issues.

However, As per my project requirement, i cannot use ESPhome. I was wondering if by any chance you have the Ardunio code handy for reading data of the inverter (PowMr POW-HVM5.5K-48V). I know I probably should start reading the protocol doc and the powerforum topic. But wanted to check first maybe you already have something! :)

Even just a basic hello world type communication example would be fine. e.g. read just one telemetry like just the Battery voltage or PV voltage would be great head start.

I want to keep the hardware part exactly as describe here. but use plain esp32 ardunio code instead of ESPHome yaml.

Thanks again.

syssi, 2024-02-03:
Sorry I cannot help here. I don't have any standalone Arduino Code handy. This project contains everything you need to know. Why do you want to implement the communication from scratch? Are you allowed to use any other library which implements the Modbus protocol?

jayprojects, 2024-02-03:
Thanks for quick reply. 
yah ardunio/c++ libraries are fine. so not exactly from scratch. Just not ESP home.

syssi, 2024-02-04:
In this case you could start here: https://docs.arduino.cc/learn/communication/modbus/

And you could try to retrieve the `ac voltage` measurement at holding register address `202` (Hex: 0xCA).

jayprojects, 2024-02-04:
Sounds good. thanks for your help.

closed 2024-02-04 by syssi