syssi/esphome-smg-ii #57: It does not work

Nando1497, 2024-05-08:
HELLO,
am I doing something wrong?
These are my logs..

[17:13:13][C][modbus_controller.switch:091]: modbus_controller.switch  Restore Mode: disabled 
[17:13:13][C][modbus_controller.switch:068]: modbus_controller.switchModbus Controller Switch 'smg-ii overload transfer to bypass enabled'
[17:13:13][C][modbus_controller.switch:091]: modbus_controller.switch  Restore Mode: disabled 
[17:13:13][C][modbus_controller.switch:068]: modbus_controller.switchModbus Controller Switch 'smg-ii battery Eq mode is enabled'
[17:13:13][C][modbus_controller.switch:091]: modbus_controller.switch  Restore Mode: disabled 
[17:13:13][C][modbus_controller.switch:068]: modbus_controller.switchModbus Controller Switch 'smg-ii remote switch'
[17:13:13][C][modbus_controller.switch:091]: modbus_controller.switch  Restore Mode: disabled 
[17:13:13][C][modbus_controller.text_sensor:012]: Modbus Controller Text Sensor 'smg-ii fault'
[17:13:13][C][modbus_controller.text_sensor:012]: Modbus Controller Text Sensor 'smg-ii warning'
[17:13:13][C][modbus_controller.text_sensor:012]: Modbus Controller Text Sensor 'smg-ii operation mode'
[17:13:13][C][sntp:055]: SNTP Time:
[17:13:13][C][sntp:056]:   Server 1: '0.pool.ntp.org'
[17:13:13][C][sntp:057]:   Server 2: '1.pool.ntp.org'
[17:13:13][C][sntp:058]:   Server 3: '2.pool.ntp.org'
[17:13:13][C][sntp:059]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[17:13:13][C][mdns:115]: mDNS:
[17:13:13][C][mdns:116]:   Hostname: smg-ii
[17:13:13][C][ota:096]: Over-The-Air Updates:
[17:13:13][C][ota:097]:   Address: smg-ii.local:3232
[17:13:13][C][ota:100]:   Using Password.
[17:13:13][C][ota:103]:   OTA version: 2.
[17:13:13][C][api:139]: API Server:
[17:13:13][C][api:140]:   Address: smg-ii.local:6053
[17:13:13][C][api:142]:   Using noise encryption: YES
[17:13:13][D][uart_debug:114]: >>> 01:03:01:4B:00:07:75:E2
[17:13:13][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=100 count=2
[17:13:13][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=108 count=2
[17:13:14][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=201 count=17
[17:13:14][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=219 count=2
[17:13:14][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=223 count=5
[17:13:14][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=331 count=7
[17:13:14][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=406 count=1
[17:13:14][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=420 count=1
[17:13:14][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=643 count=1
[17:13:14][D][modbus_controller:043]: Modbus command to device=1 register=0x14B countdown=0 no response received - removed from send queue
[17:13:14][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[17:13:14][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[17:13:14][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[17:13:14][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[17:13:15][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[17:13:15][D][modbus_controller:043]: Modbus command to device=1 register=0x64 countdown=0 no response received - removed from send queue
[17:13:15][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[17:13:15][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[17:13:15][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[17:13:16][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[17:13:16][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[17:13:16][D][modbus_controller:043]: Modbus command to device=1 register=0x6C countdown=0 no response received - removed from send queue
[17:13:16][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[17:13:16][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[17:13:17][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[17:13:17][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[17:13:17][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[17:13:17][D][modbus_controller:043]: Modbus command to device=1 register=0xC9 countdown=0 no response received - removed from send queue
[17:13:17][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[17:13:17][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[17:13:18][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[17:13:18][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[17:13:18][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[17:13:18][D][modbus_controller:043]: Modbus command to device=1 register=0xDB countdown=0 no response received - removed from send queue
[17:13:18][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[17:13:19][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[17:13:19][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[17:13:19][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[17:13:19][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[17:13:20][D][modbus_controller:043]: Modbus command to device=1 register=0xDF countdown=0 no response received - removed from send queue
[17:13:20][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[... 82 lines cut ...]
[17:13:34][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=331 count=7
[17:13:34][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=406 count=1
[17:13:34][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=420 count=1
[17:13:34][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=643 count=1
[17:13:34][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[17:13:34][D][modbus_controller:043]: Modbus command to device=1 register=0x64 countdown=0 no response received - removed from send queue
[17:13:34][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[17:13:34][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[17:13:34][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[17:13:35][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[17:13:35][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[17:13:35][D][modbus_controller:043]: Modbus command to device=1 register=0x6C countdown=0 no response received - removed from send queue
[17:13:35][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[17:13:35][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[17:13:36][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[17:13:36][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[17:13:36][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[17:13:36][D][modbus_controller:043]: Modbus command to device=1 register=0xC9 countdown=0 no response received - removed from send queue
[17:13:36][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[17:13:37][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[17:13:37][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[17:13:37][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[17:13:37][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[17:13:37][D][modbus_controller:043]: Modbus command to device=1 register=0xDB countdown=0 no response received - removed from send queue
[17:13:37][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33

Nando1497, 2024-05-13:
Yess, my yaml.

substitutions:
  name: smg-ii
  device_description: "Monitor and control a ISolar/EASUN SMG II inverter via RS232"
  tx_pin: GPIO22
  rx_pin: GPIO19

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-smg-ii"
    version: 1.1.0

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

wifi:
  ssid: "......"
  password: "......"

ota:
  password: "....."

logger:
  level: DEBUG
  # Don't write log messages to UART0 (GPIO1/GPIO3) if the inverter is connected to GPIO1/GPIO3
  baud_rate: 0

mqtt:
  broker: ......
  username: .....
  password: .....
  id: 

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt

# Enable Home Assistant API
api:
  encryption:
    key: "......"

uart:
  - id: uart_0
    baud_rate: 2400
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: smg0
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

time:
  - platform: sntp

[... 1036 lines cut ...]
    use_write_multiple: true
    address: 336
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 900.0
    step: 1
    unit_of_measurement: "min"

  # Two equalization charging intervals                           day     Uint  337 1 R/W Range: 1~90
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} two equalization charging intervals"
    use_write_multiple: true
    address: 337
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 90.0
    step: 1
    unit_of_measurement: "day"

# Button entities

  # Exit the fault mode                                           Uint  426   W   1: Exit the fault state(only when the inverter enters the fault mode , it could be available)

Nando1497, 2024-05-13:
I'm using a m5stack-atom module with RS323 
[image]

Nando1497, 2024-05-13:
I connected the following colors:
R: Orange
T: White-Orange
DC12V: Blue
G:Brown
even by inverting R with T the situation does not change

Nando1497, 2024-05-15:
[image]

Nando1497, 2024-05-15:
[image]

closed 2024-05-17 by Nando1497

syssi, 2024-05-17:
Does it work now? What did you change?