syssi/esphome-smg-ii #58: two inverters connected in parallel

amgstone, 2024-05-29:
[image]
two inverters connected in parallel. when the sun at the zenith visualizes such data, although the inverter does not show errors, it works in regular mode. It is possible to add a check in the sensors so that it does not receive incorrect data.

syssi, 2024-05-29:
Could you try to rephrase your question? I don't understand the root cause of your issue.

amgstone, 2024-06-06:
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFB
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 8341!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 80E1!=301
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 130!=00
[14:24:25][W][component:237]: Component modbus took a long time for an operation (79 ms).
[14:24:25][W][component:238]: Components should block for at most 30 ms.
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii-2 pv charging average current'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'current'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: 'measurement'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: 'A'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 1
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii-2 output voltage'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'voltage'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: ''
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: 'V'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 1
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii-2 output frequency setting'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'frequency'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: ''
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: 'Hz'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 2
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii-2 battery overvoltage protection point'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'voltage'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: ''
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: 'V'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 1
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii-2 max charging voltage'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'voltage'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: ''
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: 'V'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 1
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii-2 floating charging voltage'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'voltage'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: ''
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: 'V'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 1
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii-2 battery discharge recovery point in mains mode'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'voltage'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  State Class: ''
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Unit of Measurement: 'V'
[14:24:25][C][modbus_controller.sensor:010]: modbus_controller.sensor  Accuracy Decimals: 1
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 8341!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FBFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! E0E0!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4103!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 8541!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! E041!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! CC!=403
[14:24:25][W][modbus:108]: Modbus CRC Check failed! C071!=FA00
[14:24:25][W][modbus:108]: Modbus CRC Check failed! BC81!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FF7F
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! F845!=FFFE
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FFFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 4040!=FBFF
[14:24:25][W][modbus:108]: Modbus CRC Check failed! 42C0!=1FF
[14:24:25][W][component:237]: Component modbus took a long time for an operation (133 ms).
[14:24:25][W][component:238]: Components should block for at most 30 ms.
[14:24:26][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii-2 battery low voltage protection point in mains mode'

amgstone, 2024-06-06:
`substitutions:
  name: smg-smg-jk
  device_description: "Monitor and control two SMG II inverters via RS232 and a JK-BMS via BLE"
  smg0: "smg-ii-2"
  smg1: "smg-ii"
  tx_pin: GPIO16
  rx_pin: GPIO17
  tx_pin2: GPIO32
  rx_pin2: GPIO33

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-smg-ii"
    version: 1.1.0

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "/CygovYCtBf+QZ45PwP5RkdWiYWU9M8sszin+/1N3rA="

ota:
  password: "4a82a31dfe1dc003f3c69e73e5f12cbf"

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: "Irena_house_router"
  password: "1212121212"

  manual_ip:
    static_ip: 192.168.31.159
    gateway: 192.168.31.1
    subnet: 255.255.255.0

  ap:
    ssid: "Smg-Twin Fallback Hotspot"
    password: "YToMSNOqgNUp"

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
#    debug:
#      direction: BOTH
#      dummy_receiver: false
  - id: uart_1
    baud_rate: 9600
    tx_pin: ${tx_pin2}
    rx_pin: ${rx_pin2}
#    debug:
#      direction: BOTH
#      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

  - id: modbus1
    uart_id: uart_1
[... 1820 lines cut ...]
      }
      return values;

  # Operation Mode                                                UInt  201 1 R 0: Power On, 1: Standby, 2: Mains, 3: Off-Grid, 4: Bypass, 5: Charging, 6: Fault
  - platform: modbus_controller
    modbus_controller_id: smg1
    name: "${smg1} operation mode"
    address: 201
    register_type: holding
    raw_encode: HEXBYTES
    lambda: |-
      uint16_t value = modbus_controller::word_from_hex_str(x, 0);
      switch (value) {
        case 0: return std::string("Power On");
        case 1: return std::string("Standby");
        case 2: return std::string("Mains");
        case 3: return std::string("Off-Grid");
        case 4: return std::string("Bypass");
        case 5: return std::string("Charging");
        case 6: return std::string("Fault");
      }
      return std::string("Unknown");

    
    `

amgstone, 2024-06-06:
substitutions:
  name: smg-smg-jk
  device_description: "Monitor and control two SMG II inverters via RS232 and a JK-BMS via BLE"
  smg0: "smg-ii-2"
  smg1: "smg-ii"
  tx_pin: GPIO16
  rx_pin: GPIO17
  tx_pin2: GPIO32
  rx_pin2: GPIO33

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-smg-ii"
    version: 1.1.0

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "/CygovYCtBf+QZ45PwP5RkdWiYWU9M8sszin+/1N3rA="

ota:
  password: "4a82a31dfe1dc003f3c69e73e5f12cbf"

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: "Irena_house_router"
  password: "1212121212"

  manual_ip:
    static_ip: 192.168.31.159
    gateway: 192.168.31.1
    subnet: 255.255.255.0

  ap:
    ssid: "Smg-Twin Fallback Hotspot"
    password: "YToMSNOqgNUp"

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
#    debug:
#      direction: BOTH
#      dummy_receiver: false
  - id: uart_1
    baud_rate: 9600
    tx_pin: ${tx_pin2}
    rx_pin: ${rx_pin2}
#    debug:
#      direction: BOTH
#      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

  - id: modbus1
    uart_id: uart_1
[... 1817 lines cut ...]
        if (!values.empty()) {
          values.pop_back();
        }
      }
      return values;

  # Operation Mode                                                UInt  201 1 R 0: Power On, 1: Standby, 2: Mains, 3: Off-Grid, 4: Bypass, 5: Charging, 6: Fault
  - platform: modbus_controller
    modbus_controller_id: smg1
    name: "${smg1} operation mode"
    address: 201
    register_type: holding
    raw_encode: HEXBYTES
    lambda: |-
      uint16_t value = modbus_controller::word_from_hex_str(x, 0);
      switch (value) {
        case 0: return std::string("Power On");
        case 1: return std::string("Standby");
        case 2: return std::string("Mains");
        case 3: return std::string("Off-Grid");
        case 4: return std::string("Bypass");
        case 5: return std::string("Charging");
        case 6: return std::string("Fault");
      }
      return std::string("Unknown");

amgstone, 2024-06-06:
errors appear when the maximum production in the sun

amgstone, 2024-06-06:
[smg_new.txt](https://github.com/user-attachments/files/15614526/smg_new.txt)

syssi, 2024-06-06:
I cannot help here. This doesn't look like an software issue. It's more about bad grounding, leakage current & interferences.

amgstone, 2024-06-09:
# Average mains power                                  1W       Int   204 1 R
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${smg0} average mains power"
    address: 204
    register_type: holding
    value_type: S_WORD
    unit_of_measurement: "W"
    device_class: power
    state_class: measurement
    accuracy_decimals: 0

tell me what should be added so that it does not take a negative value

-------------------------------------------------
  # Output active power                                  1W       Int   213 1 R
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${smg0} output active power"
    address: 213
    register_type: holding
    value_type: S_WORD
    unit_of_measurement: "W"
    device_class: power
    state_class: measurement
    accuracy_decimals: 0

tell me what to write in order to add which range to accept, for example from 0 to 10000 watts

syssi, 2024-06-09:
You could filter some values:

```
```
  # Average mains power                                  1W       Int   204 1 R
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${smg0} average mains power"
    address: 204
    register_type: holding
    value_type: S_WORD
    unit_of_measurement: "W"
    device_class: power
    state_class: measurement
    accuracy_decimals: 0
    filters:
      - clamp:
          min_value: 0
          max_value: 10000
          ignore_out_of_range: true

# ...

  # Output active power                                  1W       Int   213 1 R
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${smg0} output active power"
    address: 213
    register_type: holding
    value_type: S_WORD
    unit_of_measurement: "W"
    device_class: power
    state_class: measurement
    accuracy_decimals: 0
    filters:
      - clamp:
          min_value: 0
          max_value: 10000
          ignore_out_of_range: true
```

closed 2024-07-25 by syssi