syssi/esphome-smg-ii #65: Nothing Works

cyclemat, 2024-09-02:
Have here an PowMR 5.5 K - Pinverter try to readout the inverter have all dubble checked but all i have is 
`er:183]: Duplicate modbus command found: type=0x3 address=420 count=1[0m
02.09.2024 10:43:00(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=406 count=1[0m
02.09.2024 10:43:00(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=331 count=7[0m
02.09.2024 10:43:00(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=329 count=1[0m
02.09.2024 10:43:00(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=323 count=5[0m
02.09.2024 10:43:00(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=320 count=2[0m
02.09.2024 10:42:59(-0.04 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=313 count=1[0m
02.09.2024 10:42:59(-0.05 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=201 count=17[0m
02.09.2024 10:42:50(-9.91 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=643 count=1[0m
02.09.2024 10:42:50(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=305 count=6[0m
02.09.2024 10:42:50(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=300 count=4[0m
02.09.2024 10:42:50(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=232 count=3[0m
02.09.2024 10:42:50(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=229 count=1[0m
02.09.2024 10:42:49(-0.04 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=223 count=5[0m
02.09.2024 10:42:49(-0.05 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=219 count=2[0m
02.09.2024 10:42:40(-9.9 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=420 count=1[0m
02.09.2024 10:42:40(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=406 count=1[0m
02.09.2024 10:42:40(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=331 count=7[0m
02.09.2024 10:42:40(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=329 count=1[0m
02.09.2024 10:42:40(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=323 count=5[0m
02.09.2024 10:42:40(-0 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=320 count=2[0m
02.09.2024 10:42:39(-0.04 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=313 count=1[0m
02.09.2024 10:42:39(-0.05 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=108 count=2[0m
02.09.2024 10:42:30(-9.91 seconds) 
[0;33m[W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=643 count=1[0m`

[image]

can anybody help me

syssi, 2024-09-02:
Please increase the log level to `DEBUG` and provide another log: https://github.com/syssi/esphome-smg-ii?tab=readme-ov-file#debugging

Could you take some photos of your wiring and RS485 module? I guess your wiring isn't correct or the flow control is not properly managed.

cyclemat, 2024-09-02:
RS485 ? in the readme stands RS232 !

Here are Pictures will set the debug level this evening ! 

[image]
[image]
[image]
[image]

syssi, 2024-09-02:
> RS485 ? in the readme stands RS232 !

My fault. It's RS232! :-)

syssi, 2024-09-02:
Could you check the chip model of your RS232 converter? It's important to use the `MAX3232` (3.3V model) and no `MAX232` (5V model).

cyclemat, 2024-09-02:
Will check but i have buy IT as 3,3v habe ordered new ones to test 

https://www.ebay.de/itm/255312120055?mkcid=16&mkevt=1&mkrid=707-127634-2357-0&ssspo=kSLcqpXwSAK&sssrc=4429486&ssuid=0HguCZJXT4u&var=&widget_ver=artemis&media=COPY

This is What is in use in the moment
 https://www.ebay.de/itm/255303534424?mkcid=16&mkevt=1&mkrid=707-127634-2357-0&ssspo=kSLcqpXwSAK&sssrc=4429486&ssuid=0HguCZJXT4u&var=&widget_ver=artemis&media=COPY

cyclemat, 2024-09-02:
mmmmh 
[image]

cyclemat, 2024-09-02:
and is this normal in compiling ? 
[image]

cyclemat, 2024-09-04:
so i have rebuild all with a new ESP and a new MAX2322 and had to need swap the RX and TX pin and now it works look mine ESP32 has a wrong pinout  thanks for help

closed 2024-09-04 by cyclemat