syssi/esphome-smg-ii #66: Daxtromn AGH-10.2KW (analog VICTOR NM-MAX-10.2KWs) support

fastvd, 2024-09-03:
hello everyone
I have such an inverter at home, it works with a Wi-Fi adapter and the smartESS application ... I am trying to integrate it into my home assistant through the esp32 board ... I understand that this is not a PIP inverter (pipsolar tried, but nothing worked either), so it is long and boring killing my time, I realized that I was given that will give me another firmware...
the cable is 100% working, because I have a response from the inverter in the logs and the light is blinking on my rsr 32-ttl ... the question is what am I doing wrong) here are my logs when I turned on advanced logging:
my conf:
substitutions:
  name: smg-ii
  device_description: "Monitor and control a ISolar/EASUN SMG II inverter via RS232"
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-smg-ii"
    version: 1.2.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 10.12.17.173
    # Set this to the IP address of the router. Often ends with .1
    gateway: 10.12.17.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

ota:
  - platform: esphome
    password: "a3f14afd0bda14550c0ac1bfc4499eee"

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

# api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: smg0
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
[... 362 lines cut ...]
[18:49:34][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:40:C7:FE:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:40:D0:FC:FF:FF:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:CD:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii battery average power':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_battery_average_power/state'
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii pv average voltage':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_pv_average_voltage/state'
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii pv average current':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_pv_average_current/state'
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii pv average power':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_pv_average_power/state'
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii pv charging average power':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_pv_charging_average_power/state'
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii load percentage':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_load_percentage/state'
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii dcdc temperature':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_dcdc_temperature/state'
[18:49:34][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii inverter temperature':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_inverter_temperature/state'
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii battery state of charge':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_battery_state_of_charge/state'
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii battery average current':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_battery_average_current/state'
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii inverter charging average current':
[18:49:34][C][mqtt.sensor:032]:   State Topic: 'smg-ii/sensor/smg-ii_inverter_charging_average_current/state'
[18:49:34][C][mqtt.sensor:028]: MQTT Sensor 'smg-ii pv charging average current':

syssi, 2024-09-03:
The responses look like garbage only / a wrong baud rate. Do you know any protocol details  of your inverter?

fastvd, 2024-09-05:
https://daxtromn-power.com/blogs/solar-inverter-solution/daxtromn-power-solar-inverter-communication-protocol

fastvd, 2024-09-05:
again:
1) I know for sure that the program on the smartess phone works stably, but slowly
2) I know for sure that through usb-rs232 + windows + watchPower I also receive data... at this moment COM port 9600 8bit 1 stop
3) if I even get this garbage, does it mean that the physical connection is working?

syssi, 2024-09-05:
If you have two ESPs and two RS232 you could build a test setup (one transmitter + one receiver). I will provide all required details. IMO your wiring/RS232 converter will be the root cause.

fastvd, 2024-09-05:
I even have 3 sets)

syssi, 2024-09-05:
Please flash these configuration to your ESPs:

1. device: https://raw.githubusercontent.com/syssi/esphome-smg-ii/main/tests/esp32-uart-trx1.yaml
2. device: https://raw.githubusercontent.com/syssi/esphome-smg-ii/main/tests/esp32-uart-trx2.yaml

Connect `GPIO16` (TX) of the first ESP to `GPIO17` (RX) of the second and vice versa plus GND to GND.

```
        ┌──────────┐                ┌─────────┐
        │        16│<-------------->│17       │
        │  First 17│<-------------->│16       │
        │  ESP32   │<----- GND ---->│  Second │
VCC --->│          │                │    ESP  │<--- VCC
GND --->│          │                │         │<--- GND
        └──────────┘                └─────────┘
```

Take a look at the log of the ESPHome nodes. Do you see incoming and outgoing traffic?

``` 
>>> "Hello from uart-trx1"
<<< "Hello from uart-trx2"
```

As next step we will add two RS232 converters between the ESPs.

fastvd, 2024-09-05:
:37:22][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:22][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:23][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:23][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:24][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:24][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:25][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:25][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:26][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:26][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:27][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:27][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:28][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:28][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:29][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:29][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:30][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:30][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:31][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:31][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:32][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:32][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:33][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32
[16:37:33][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:75:61:72:74:2D:74:72:78:32

fastvd, 2024-09-05:
identical picture on 2 boards

fastvd, 2024-09-05:
as far as I understand, the boards are live

syssi, 2024-09-05:
Die you flash the same YAML to both ESPs!?

fastvd, 2024-09-05:
**substitutions:
  name: uart-trx1
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: rx
  friendly_name: rx

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  - platform: esphome
    password: "c796c75866b9715fc5a9946dd3e9cf05"

logger:
  level: DEBUG

api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: true

interval:
  - interval: 1s
    then:
      - uart.write:
          id: uart_0
          data: "Hello from RX"**

fastvd, 2024-09-05:
substitutions:
  name: uart-trx2
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: tx
  friendly_name: tx

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: true

interval:
  - interval: 1s
    then:
      - uart.write:
          id: uart_0
          data: "Hello from TX"

fastvd, 2024-09-05:
[17:57:38][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[17:57:39][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[17:57:39][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[17:57:40][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[17:57:40][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[17:57:41][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[17:57:41][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[17:57:42][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[17:57:42][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[17:57:43][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[17:57:43][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[17:57:44][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[17:57:44][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58

fastvd, 2024-09-05:
52:58 and 54:58 its "RX" and "TX" if I understand correctly

syssi, 2024-09-05:
That's perfect! Both ESPs are healthy. Next step is to transform the UART-TTL (logic level 3.3V) to RS232 (logic level -12/+12V) and back using the second RS232 converter down to the logic level of 3.3V. 

```
        ┌──────────┐            ┌─────────┐       ┌─────────┐            ┌─────────┐
        │        16│---- TX --->│       TX│<----->│RX       │<--- TX ----│16       │
        │  First 17│<--- RX ----│       RX│<----->│TX       │---- RX --->│17       │
        │  ESP32   │<--- GND -->│      GND│<----->│GND      │<--- GND -->│  Second │
VCC --->│          │<-- 3.3V -->│         │       │         │<-- 3.3V -->│    ESP  │<--- VCC
GND --->│          │            │ MAX3232 │       │ MAX3232 │            │         │<--- GND
        └──────────┘            └─────────┘       └─────────┘            └─────────┘
```

This setup is super easy to mess up because it's super likely to swap RX/TX accidently.

fastvd, 2024-09-06:
02][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:03][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:03][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:04][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:04][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:05][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:05][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:06][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:06][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:07][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:07][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:08][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:08][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:09][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:09][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:10][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:10][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:11][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:11][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:12][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:12][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:13][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:13][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:14][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:14][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:15][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:15][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:16][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:16][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:17][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:17][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:18][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:18][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58
[13:23:19][D][uart_debug:114]: <<< 48:65:6C:6C:6F:20:66:72:6F:6D:20:54:58
[13:23:19][D][uart_debug:114]: >>> 48:65:6C:6C:6F:20:66:72:6F:6D:20:52:58

fastvd, 2024-09-06:
3:44:06][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:06][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[13:44:06][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:06][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[13:44:06][D][modbus_controller:040]: Modbus command to device=1 register=0xDF countdown=0 no response received - removed from send queue
[13:44:06][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:06][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[13:44:06][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[13:44:07][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[13:44:07][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[13:44:07][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[13:44:07][D][modbus_controller:040]: Modbus command to device=1 register=0xE5 countdown=0 no response received - removed from send queue
[13:44:07][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[13:44:07][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:08][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[13:44:08][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:08][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[13:44:08][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:08][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[13:44:08][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:08][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[13:44:08][D][modbus_controller:040]: Modbus command to device=1 register=0x196 countdown=0 no response received - removed from send queue
[13:44:08][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:08][D][uart_debug:114]: >>> 01:03:01:A4:00:01:C4:15
[13:44:09][D][uart_debug:114]: >>> 01:03:01:A4:00:01:C4:15
[13:44:09][D][uart_debug:114]: >>> 01:03:01:A4:00:01:C4:15
[13:44:09][D][uart_debug:114]: >>> 01:03:01:A4:00:01:C4:15
[13:44:09][D][uart_debug:114]: >>> 01:03:01:A4:00:01:C4:15
[13:44:09][D][modbus_controller:040]: Modbus command to device=1 register=0x1A4 countdown=0 no response received - removed from send queue
[13:44:10][D][uart_debug:114]: >>> 01:03:02:83:00:01:74:5A
[13:44:10][D][uart_debug:114]: >>> 01:03:02:83:00:01:74:5A
[13:44:10][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:10][D][uart_debug:114]: >>> 01:03:02:83:00:01:74:5A
[13:44:10][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:10][D][uart_debug:114]: >>> 01:03:02:83:00:01:74:5A
[13:44:10][D][uart_debug:114]: <<< 00:78:80:80:78:FC
[13:44:10][D][uart_debug:114]: >>> 01:03:02:83:00:01:74:5A
[13:44:11][D][modbus_controller:040]: Modbus command to device=1 register=0x283 countdown=0 no response received - removed from send queue
[13:44:11][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:11][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[13:44:11][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[13:44:11][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[13:44:11][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[13:44:12][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[13:44:12][D][modbus_controller:040]: Modbus command to device=1 register=0x64 countdown=0 no response received - removed from send queue
[13:44:12][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[13:44:12][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[13:44:12][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[13:44:12][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=232 count=3
[13:44:12][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=300 count=4
[13:44:12][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=305 count=6
[13:44:12][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=313 count=1
[13:44:12][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=320 count=2
[13:44:12][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=323 count=5
[13:44:12][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=329 count=1
[13:44:12][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=331 count=7
[13:44:12][W][component:237]: Component modbus_controller took a long time for an operation (65 ms).
[13:44:12][W][component:238]: Components should block for at most 30 ms.
[13:44:12][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[13:44:13][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[13:44:13][D][modbus_controller:040]: Modbus command to device=1 register=0xE8 countdown=0 no response received - removed from send queue
[13:44:13][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[13:44:13][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[13:44:13][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[13:44:14][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[13:44:14][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[13:44:14][D][modbus_controller:040]: Modbus command to device=1 register=0x12C countdown=0 no response received - removed from send queue
[13:44:14][D][uart_debug:114]: >>> 01:03:01:31:00:06:95:FB
[13:44:14][D][uart_debug:114]: >>> 01:03:01:31:00:06:95:FB
[13:44:14][D][uart_debug:114]: >>> 01:03:01:31:00:06:95:FB
[... 8 lines cut ...]
[... cut ...]
[13:44:18][D][uart_debug:114]: <<< 00:78:80:80:78:00:78:78:78:80:78:80:78:F8:00
[13:44:19][D][uart_debug:114]: >>> 01:03:01:49:00:01:54:20
[13:44:19][D][uart_debug:114]: >>> 01:03:01:49:00:01:54:20
[13:44:19][D][uart_debug:114]: >>> 01:03:01:49:00:01:54:20
[13:44:19][D][uart_debug:114]: >>> 01:03:01:49:00:01:54:20
[13:44:19][D][uart_debug:114]: >>> 01:03:01:49:00:01:54:20
[13:44:20][D][modbus_controller:040]: Modbus command to device=1 register=0x149 countdown=0 no response received - removed from send queue
[13:44:20][D][uart_debug:114]: >>> 01:03:01:4B:00:07:75:E2

fastvd, 2024-09-06:
from esp32 to esp32 via rs232, the connection is already ok.. I connected it to the inverter.. I see that there is also an answer, but it's still rubbish... maybe I still have PIP?

fastvd, 2024-09-06:
][D][sensor:094]: 'pipsolar grid_voltage': Sending state 239.30000 V with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar grid_frequency': Sending state 49.90000 Hz with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar ac_output_voltage': Sending state 219.80000 V with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar ac_output_frequency': Sending state 49.90000 Hz with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar ac_output_apparent_power': Sending state 549.00000 VA with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar ac_output_active_power': Sending state 410.00000 W with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar output_load_percent': Sending state 5.00000 % with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar bus_voltage': Sending state 440.00000 V with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar battery_voltage': Sending state 54.90000 V with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar battery_charging_current': Sending state 46.00000 A with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar battery_capacity_percent': Sending state 67.00000 % with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar inverter_heat_sink_temperature': Sending state 37.00000 °C with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar pv_input_current_for_battery': Sending state 9.30000 A with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar pv_input_voltage': Sending state 377.79999 V with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar battery_voltage_offset_for_fans_on': Sending state 0.00000 V with 1 decimals of accuracy
[15:39:22][D][sensor:094]: 'pipsolar pv_charging_power': Sending state 3450.00000 W with 1 decimals of accuracy
[15:39:22][W][component:237]: Component esphome.coroutine took a long time for an operation (101 ms).

fastvd, 2024-09-06:
after all, I have a PIP inverter... so if it works through the SmartESS application, it is not a fact that the inverter is not PIP!
Thank you very much for your help!

P.S.: I had a problem with the pinout...namely, the TX from the ESP32 board should go to the TX of the rs232-ttl board...

syssi, 2024-09-06:
There is  SmartESS dongle for PIP and a special version for Modbus communication.

closed 2024-09-06 by syssi

Samuele92, 2025-03-17:
> ciao a tutti ho un inverter del genere a casa, funziona con un adattatore Wi-Fi e l'applicazione smartESS... sto cercando di integrarlo nel mio assistente domestico tramite la scheda esp32... ho capito che questo non è un inverter PIP (pipsolar ha provato, ma non ha funzionato neanche lui), quindi è lungo e noioso e mi fa perdere tempo, ho capito che mi è stato dato che mi darà un altro firmware... il cavo funziona al 100%, perché ho una risposta dall'inverter nei log e la luce lampeggia sul mio rsr 32-ttl... la domanda è cosa sto sbagliando) ecco i miei log quando ho attivato la registrazione avanzata: _la mia conf:_ **sostituzioni: nome: smg-ii device_description: "Monitoraggio e controllo di un inverter ISolar/EASUN SMG II tramite RS232" tx_pin: GPIO16 rx_pin: GPIO17
> 
> esphome: nome: ${name} commento: ${device_description} min_version: 2024.6.0 progetto: nome: "syssi.esphome-smg-ii" versione: 1.2.0
> 
> esp32: scheda: esp32dev framework: tipo: esp-idf
> 
> wifi: ssid: !segreto wifi_ssid password: !segreto wifi_password manual_ip: static_ip: 10.12.17.173 gateway: 10.12.17.1 subnet: 255.255.255.0
> 
> oppure:
> 
> * piattaforma: esphome 
>   password: "a3f14afd0bda14550c0ac1bfc4499eee"
> 
> logger: livello: DEBUG
> 
> mqtt: broker: !secret mqtt_host nome utente: !secret mqtt_username password: !secret mqtt_password id: mqtt_client
> 
> parte:
> 
> * id: uart_0 
>   baud_rate: 9600 
>   tx_pin: ${tx_pin} 
>   rx_pin: ${rx_pin} 
>   debug: 
>   direzione: ENTRAMBI 
>   dummy_receiver: falso
> 
> modbus:
> 
> * id: modbus0 
>   uart_id: uart_0 
>   tempo_di_attesa_invio: 200 ms
> 
> regolatore_modbus:
> 
> * id: smg0 
>   indirizzo: 0x01 
>   modbus_id: modbus0 
>   command_throttle: 200ms 
>   update_interval: 10s
> 
> tempo:
> 
> * piattaforma: sntp
> 
> sensore:
> 
> * piattaforma: total_daily_energy 
>   nome: "${name} energia FV oggi" 
>   ripristino: vero 
>   icona: mdi:counter 
>   power_id: smg0_pv_average_power 
>   filtri:
>   
>   * moltiplica: 0,001 
>     unità_di_misura: kWh
> * piattaforma: modbus_controller 
>   modbus_controller_id: smg0 
>   nome: "${name} codice errore" 
>   indirizzo: 100 
>   register_type: holding 
>   value_type: U_DWORD 
>   precision_decimals: 0
> * piattaforma: modbus_controller 
>   modbus_controller_id: smg0 
>   nome: "codice di avviso ${name}" 
>   indirizzo: 108 
>   tipo_registro: in attesa 
>   tipo_valore: U_DWORD 
>   decimali_accuratezza: 0**
> 
> ...... quindi la configurazione standard**
> 
> _e i miei registri:_
> 
> **18:49:31][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30 [18:49:31][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:31][W][modbus:114]: Controllo CRC Modbus fallito! A17E!=00 [18:49:31][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:31][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:31][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:31][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:31][W][modbus:114]: Controllo CRC Modbus fallito! 8055!=FFFF [18:49:31][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:32][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:32][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:32][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:32][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:32][W][modbus:114]: Controllo CRC Modbus fallito! C071!=00 [18:49:32][W][modbus:114]: Controllo CRC Modbus fallito! AA56!=00 [18:49:32][W][component:237]: Il componente modbus ha impiegato molto tempo per un'operazione (82 ms). [18:49:32][W][component:238]: I componenti dovrebbero bloccarsi per al massimo 30 ms. [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensorSensore del controller Modbus 'smg-ii operation mode id' [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensor Classe di stato: '' [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensor Unità di misura: '' [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensor Decimali di precisione: 0 [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensorSensore del controller Modbus 'smg-ii ac voltage' [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensor Classe dispositivo: 'tensione' [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensor Classe stato: 'misurazione' [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensor Unità di misura: 'V' [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensor Decimali di precisione: 1 [18:49:32][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:10:00:FC:FF:FF:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:8E:7F:FF:FF:00:00:00:00:00:00:00:00:00 :00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:20:00:80:E1:FF:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00 [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensorSensore del controller Modbus 'smg-ii ac frequency' [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensor Classe dispositivo:'frequenza' [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensor Classe di stato: 'misurazione' [18:49:32][C][modbus_controller.sensor:010]: modbus_controller.sensor Unità di misura: 'Hz' [18:49:32][C][modbus_controller.sensor:010]: modbu [... cut ...] battery_state_of_charge/state' [18:49:34][C][mqtt.sensor:028]: Sensore MQTT 'corrente media della batteria smg-ii': [18:49:34][C][mqtt.sensor:032]: Argomento di stato: 'smg-ii/sensor/smg-ii_battery_average_current/state' [18:49:34][C][mqtt.sensor:028]: Sensore MQTT 'smg-ii inverter carica media corrente': [18:49:34][C][mqtt.sensor:032]: Argomento di stato: 'smg-ii/sensor/smg-ii_inverter_carica_media_corrente/stato' [18:49:34][C][mqtt.sensor:028]: Sensore MQTT 'smg-ii pv carica media corrente':**

Samuele92, 2025-03-17:
Hi, can I have the yaml compilation of the Daxtromn AGH-10.2KW inverter? Thanks