syssi/esphome-smg-ii #68: Inverter SMG II Easun problem connection

lxdr23, 2024-09-04:
### Discussed in https://github.com/syssi/esphome-smg-ii/discussions/67

<div type='discussions-op-text'>

<sup>Originally posted by **lxdr23** September  4, 2024</sup>
Hi,  this inverter I am trying to configure the esphome-smg-ii package via esphome from homeassistant, I directly inserted the content of the esp8266-example.yaml file into esphome and programmed a wemos d1 mini with all the sensors displayed in HA. However, the connection does not occur, am I doing the wiring wrong?

thanks 1000

[image]
[image]
</div>

lxdr23, 2024-09-04:
the full code:   

substitutions:
  name: smg-ii
  device_description: "Monitor and control a ISolar/EASUN SMG II inverter via RS232"
  tx_pin: GPIO4
  rx_pin: GPIO5

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-smg-ii"
    version: 1.2.0

esp8266:
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt

api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: smg0
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

time:
  - platform: sntp

sensor:
  - platform: total_daily_energy
    name: "${name} PV energy today"
    restore: true
    icon: mdi:counter
    power_id: smg0_pv_average_power
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
    unit_of_measurement: kWh

  # Fault code                                                    ULong 100 2 R
[... 1024 lines cut ...]
    use_write_multiple: true
    address: 336
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 900.0
    step: 1
    unit_of_measurement: "min"

  # Two equalization charging intervals                           day     Uint  337 1 R/W Range: 1~90
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} two equalization charging intervals"
    use_write_multiple: true
    address: 337
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 90.0
    step: 1
    unit_of_measurement: "day"

# Button entities

  # Exit the fault mode                                           Uint  426   W   1: Exit the fault state(only when the inverter enters the fault mode , it could be available)

syssi, 2024-09-04:
I've deleted the discussion because we don't need to discuss the topic on two places. 

Is your ESP8266 / ESPHome node stable and are you able to connect to the node and retrieve a log book?

lxdr23, 2024-09-04:
yes:   INFO ESPHome 2024.8.3
INFO Reading configuration /config/esphome/esphome-web-99a222.yaml...
INFO Detected timezone 'Europe/Rome'
INFO Starting log output from 192.168.10.52 using esphome API
INFO Successfully connected to smg-ii @ 192.168.10.52 in 0.005s
WARNING Can't connect to ESPHome API for smg-ii @ 192.168.10.52: Timeout waiting for HelloResponse, ConnectResponse after 30.0s (TimeoutAPIError)
INFO Trying to connect to smg-ii @ 192.168.10.52 in the background
WARNING smg-ii @ 192.168.10.52: Connection error occurred: smg-ii @ 192.168.10.52: Connection lost
INFO Successfully connected to smg-ii @ 192.168.10.52 in 0.005s
INFO Successful handshake with smg-ii @ 192.168.10.52 in 0.030s
[16:35:33][I][app:100]: ESPHome version 2024.8.3 compiled on Sep  4 2024, 12:26:10
[16:35:33][I][app:102]: Project syssi.esphome-smg-ii version 1.2.0
[16:35:40][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=305 count=6
[16:35:40][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=313 count=1
[16:35:40][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=323 count=5
[16:35:40][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=331 count=7
[16:35:40][W][component:237]: Component modbus_controller took a long time for an operation (60 ms).
[16:35:50][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=108 count=2
[16:35:50][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=201 count=17
[16:35:50][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=223 count=5
[16:35:50][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=232 count=3
[16:35:50][W][component:237]: Component modbus_controller took a long time for an operation (60 ms).
[16:36:00][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=305 count=6
[16:36:00][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=313 count=1
[16:36:00][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=323 count=5
[16:36:00][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=331 count=7
[16:36:00][W][component:237]: Component modbus_controller took a long time for an operation (60 ms).
WARNING smg-ii @ 192.168.10.52: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for smg-ii @ 192.168.10.52
WARNING Disconnected from API
INFO Successfully connected to smg-ii @ 192.168.10.52 in 0.004s
WARNING smg-ii @ 192.168.10.52: Connection error occurred: smg-ii @ 192.168.10.52: EOF received
WARNING Can't connect to ESPHome API for smg-ii @ 192.168.10.52: smg-ii @ 192.168.10.52: EOF received (SocketClosedAPIError)
INFO Trying to connect to smg-ii @ 192.168.10.52 in the background
INFO Successfully connected to smg-ii @ 192.168.10.52 in 0.006s
WARNING smg-ii @ 192.168.10.52: Connection error occurred: smg-ii @ 192.168.10.52: EOF received

syssi, 2024-09-04:
Could you increase the log level and provide another log? See https://github.com/syssi/esphome-smg-ii?tab=readme-ov-file#debugging

Please check the model of your RS232 converter. It must be an MAX3232 (= 3.3V version). A MAX232 won't work.

lxdr23, 2024-09-04:
ok the ttl converte is this
[image]

i have set de debuh level:
[image]

and the logs are;

INFO ESPHome 2024.8.3
INFO Reading configuration /config/esphome/esphome-web-99a222.yaml...
INFO Detected timezone 'Europe/Rome'
INFO Starting log output from 192.168.10.52 using esphome API
INFO Successfully connected to smg-ii @ 192.168.10.52 in 0.007s
INFO Successful handshake with smg-ii @ 192.168.10.52 in 0.039s
[17:00:28][I][app:100]: ESPHome version 2024.8.3 compiled on Sep  4 2024, 16:54:42
[17:00:28][I][app:102]: Project syssi.esphome-smg-ii version 1.2.0
[17:00:28][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii output voltage'
[17:00:28][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii battery overvoltage protection point'
[17:00:28][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii floating charging voltage'
[17:00:28][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii battery low voltage protection point in mains mode'
[17:00:28][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii maximum charging current'
[17:00:28][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii maximum mains charging current'
[17:00:28][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii Eq Charging voltage'
[17:00:28][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii battery equalization time'
[17:00:28][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii equalization Timeout exit'
[17:00:28][C][total_daily_energy:023]: Total Daily Energy 'smg-ii PV energy today'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii fault code'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii warning code'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii operation mode id'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii ac voltage'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii average mains power'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii effective inverter current'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii average inverter power'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'power'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii output effective voltage'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'voltage'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii output effective Current'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii output frequency'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii output apparent power'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii battery average Current'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'current'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii pv average voltage'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii pv average power'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'power'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii pv charging average power'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii load percentage'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii inverter temperature'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensor  Device Class: 'temperature'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii battery average current'
[17:00:28][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii pv charging average current'
[17:00:28][C][modbus_controller.select:009]: modbus_controller.selectModbus Controller Select 'smg-ii output mode'
[17:00:28][C][modbus_controller.select:009]: modbus_controller.selectModbus Controller Select 'smg-ii output priority'
[17:00:28][C][modbus_controller.select:009]: modbus_controller.selectModbus Controller Select 'smg-ii input voltage range'
[17:00:28][C][modbus_controller.select:009]: modbus_controller.selectModbus Controller Select 'smg-ii lcd backlight'
[17:00:28][C][modbus_controller.select:009]: modbus_controller.selectModbus Controller Select 'smg-ii turn on mode'
[17:00:28][C][modbus_controller.switch:068]: modbus_controller.switchModbus Controller Switch 'smg-ii energy-saving mode'
[17:00:28][C][modbus_controller.switch:068]: modbus_controller.switchModbus Controller Switch 'smg-ii over temperature automatic restart'
[17:00:28][C][modbus_controller.switch:068]: modbus_controller.switchModbus Controller Switch 'smg-ii battery Eq mode is enabled'
[17:00:28][C][modbus_controller.switch:068]: modbus_controller.switchModbus Controller Switch 'smg-ii remote switch'
[... cut ...]
[17:00:31][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[17:00:32][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[17:00:32][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[17:00:32][D][modbus_controller:037]: Modbus command to device=1 register=0xE8 countdown=0 no response received - removed from send queue
[17:00:32][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[17:00:32][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[17:00:33][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[17:00:33][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C

cyclemat, 2024-09-06:
look your wiring not good conections look the white/orange cable use flux to weld

fastvd, 2024-09-06:
soldered to D1 and D2, and D4 and D5 are in the configuration

lxdr23, 2024-09-06:
The connection is ok but not work

lxdr23, 2024-09-06:
[image]

syssi, 2024-09-06:
@fastvd Could you help here because you are an expert in wiring this type of MAX3232 modules now. ;-) Thanks in advance!

fastvd, 2024-09-09:
here is a 100% correct TX RX
[image]

lxdr23, 2024-09-09:
Thnaks 1000, then I think my inverter is not compatible, I wired it like this
[image]

[image]

lxdr23, 2024-09-09:
I'm really desperate

syssi, 2024-12-26:
Do you still need some support here?

closed 2025-07-05 by syssi