syssi/esphome-smg-ii #69: cant write to Output priority and Battery charging priority

cyclemat, 2024-09-05:
nos i become all datas but the problem is cant change Output priority and Battery charging priority have try to send direct via mqqt and openhab but it didnt work have try it via smg-ii/select/smg-ii_output_priority/command/ its only a me problem or has anybody the same problem ?

syssi, 2024-09-05:
What did you send exactly? The topic is called `smg-ii/select/smg-ii_output_priority/command`. There is no trailing slash.

cyclemat, 2024-09-05:
have send 1,2,3 nothing happens 
[image]

syssi, 2024-09-05:
1. Please remove the trailing slash from the topic:

```
smg-ii/select/smg-ii_output_priority/command

vs. 

smg-ii/select/smg-ii_output_priority/command/
```

2. Select `raw` instead of `json`
3. Send a value between 0 and 3.

```
      "Utility-PV-Battery (UTI)": 0
      "PV-Utility-Battery (SOL)": 1
      "PV-Battery-Utility (SBU)": 2
      "PV-Utility-Battery (SUB)": 3
```

cyclemat, 2024-09-05:
[image]
[image]
nothing happens

syssi, 2024-09-05:
Hmm... this is new to me and probably a bug or incompatibility. If you change the `output priority` using the on screen display does the state of the number entity change? In other words: Does reading the current state of the `output priority` work and are you able to change the mode in general?

syssi, 2024-09-05:
Could you try to transmit the label/option name instead of the integer?

```
```
$ mosquitto_pub 'smg-ii/select/smg-ii_output_priority/command' -m 0

[10:12:18][D][select:062]: 'smg-ii output priority' - Setting
[10:12:18][W][select:111]: 'smg-ii output priority' - Option 0 is not a valid option

$ mosquitto_pub -h 192.168.1.100 -u wilfried -P oc-0p1um -t 'smg-ii/select/smg-ii_output_priority/command' -m "Utility-PV-Battery (UTI)"
[10:13:10][D][select:062]: 'smg-ii output priority' - Setting
[10:13:10][D][select:115]: 'smg-ii output priority' - Set selected option to: Utility-PV-Battery (UTI)
[10:13:10][D][modbus_controller.select:048]: Found value 0 for option 'Utility-PV-Battery (UTI)'
```

cyclemat, 2024-09-05:
have test Utility-PV-Battery (UTI) and it works
[image]

syssi, 2024-09-05:
Could you provide a valid `maximum charging current` value of your inverter? I would like to add some examples to the YAML like:

```
# If you use `mqtt` you can control the inverter using the `command` topics. The topic depends on
# the name of your ESPHome node and the name of the entity:
#
# mosquitto_pub -t 'smg-ii/select/smg-ii_output_priority/command' -m "Utility-PV-Battery (UTI)"
# mosquitto_pub -t 'smg-ii/switch/smg-ii_energy-saving_mode/command' -m 'ON'    # Supported values: ON, OFF, TOGGLE
# mosquitto_pub -t 'smg-ii/number/smg-ii_maximum_charging_current/command' -m 10

```

cyclemat, 2024-09-06:
its not a bug it a feature ;) thanks

closed 2024-09-06 by cyclemat

syssi, 2024-09-06:
@cyclemat Did you miss my last question?

cyclemat, 2024-09-06:
Mmmh Max charging for Utility charging or complete ?

Utility charging from 2a up
For conplete 5a up