syssi/esphome-smg-ii #71: Config "esp8266-example.yaml" makes an esp8266 into botloop by some options

trianglesis, 2024-09-07:
Good day.

I connected everything and made it work just fine for ALL sensors.
However, some parts of the config right after the line cause my dongle to enter a continuous boot loop.

```
select:
  # Output Mode                                                   Uint  300 1 R/W 0: Single, 1: Parallel, 2: 3 Phase-P1, 3: 3 Phase-P2, 4: 3 Phase-P3
  - platform: modbus_controller
```

I only understand there is an issue after I ran Flash from HA's ESPHome UI by adding only a few sensors one by one.

After reproducing this issue a few times, I also noticed that after I add just a single `select` item—the `output mode` dropdown—the board reboots 2-3 times and becomes stable.

But as I continue adding more sections: `switch,` `text_sensor,` or `number` - the board goes wild and only appears online for 2-5 seconds. The board continues to run in a boot loop until I forcefully re-flash it with the fresh firmware:

- https://web.esphome.io/

Here, it works fine with only a single element, "# Output Mode," and all sensors from above.

[image]

I can't get meaningful logs from this failed state since it only comes to life for 5 seconds.
In USB mode, it also fires a ton of CRC-related errors when wiping the console, but I'll try in successive iterations.

My inverter:
- ISolar-SMG-II-4KW-24V WiFi

My board:

[image]

Later I'll add more logs if I'll be able to collect them.

trianglesis, 2024-09-07:
After I've added only one item:

```yaml
######
# This section was added during the previous flash and works fine after 2-3 reboots:
######
select:
  # Output Mode                                                   Uint  300 1 R/W 0: Single, 1: Parallel, 2: 3 Phase-P1, 3: 3 Phase-P2, 4: 3 Phase-P3
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} output mode"
    use_write_multiple: true
    address: 300
    value_type: U_WORD
    optionsmap:
      "Single": 0
      "Parallel": 1
      "Phase P1": 2
      "Phase P2": 3
      "Phase P3": 4

######
# Added this recently
######
  # Output priority                                               Uint  301 1 R/W 0: Utility-PV-Battery, 1: PV-Utility-Battery, 2: PV-Battery-Utility
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${name} output priority"
    use_write_multiple: true
    address: 301
    value_type: U_WORD
    optionsmap:
      "Utility-PV-Battery (UTI)": 0
      "PV-Utility-Battery (SOL)": 1
      "PV-Battery-Utility (SBU)": 2
      "PV-Utility-Battery (SUB)": 3
```

Log:

```log
[13:11:48][E][ota.arduino_esp8266:036]: Begin error: 14
[13:11:48][E][component:164]: Component esphome.ota set Error flag: unspecified
[13:11:53][E][component:176]: Component esphome.ota cleared Error flag
[13:12:50][E][json:041]: Could not allocate memory for JSON document! Requested 496 bytes, largest free heap block: 496 bytes
```

Flash log:
```log
INFO ESPHome 2024.8.3
INFO Reading configuration /config/esphome/esphome-web-64e0e5.yaml...
INFO Detected timezone 'Europe/Kyiv'
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-64e0e5 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
Compiling .pioenvs/esphome-web-64e0e5/src/main.cpp.o
Linking .pioenvs/esphome-web-64e0e5/firmware.elf
RAM:   [=====     ]  46.6% (used 38136 bytes from 81920 bytes)
Flash: [=====     ]  49.2% (used 503569 bytes from 1023984 bytes)
Building .pioenvs/esphome-web-64e0e5/firmware.bin
esp8266_copy_factory_bin([".pioenvs/esphome-web-64e0e5/firmware.bin"], [".pioenvs/esphome-web-64e0e5/firmware.elf"])
esp8266_copy_ota_bin([".pioenvs/esphome-web-64e0e5/firmware.bin"], [".pioenvs/esphome-web-64e0e5/firmware.elf"])
========================= [SUCCESS] Took 5.21 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.105
INFO Uploading /data/build/esphome-web-64e0e5/.pioenvs/esphome-web-64e0e5/firmware.bin (507728 bytes)
INFO Compressed to 348128 bytes
ERROR Error binary size: Unknown error from ESP
```

trianglesis, 2024-09-07:
I've skipped the `select` and `switch` sections entirely.
Now, adding this first only:

```yaml
text_sensor:
  # Fault code                                                    ULong 100 2 R
  - platform: modbus_controller
```

And the board goes wild again, losing a wireless connection. I'm going into USB.

```log
[13:30:27][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:27][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:27][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:28][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:28][I][wifi:540]: - 'SSID_NAME!' (62:48:B8:42:0C:CC) [redacted]▂▄▆█
[13:30:28][I][wifi:313]: WiFi Connecting to 'SSID_NAME!'...
[13:30:28][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:28][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:29][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:29][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:29][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=232 count=3
[13:30:29][W][modbus_controller:182]: Duplicate modbus command found: type=0x3 address=643 count=1
[13:30:30][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:30][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:30][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:30][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:30][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:31][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:31][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:31][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:31][I][wifi:617]: WiFi Connected!
[13:30:31][W][component:157]: Component mqtt set Warning flag: unspecified
[13:30:31][I][mqtt:244]: Connecting to MQTT...
[13:30:31][W][component:170]: Component wifi cleared Warning flag
[13:30:31][W][component:170]: Component mqtt cleared Warning flag
[13:30:31][I][mqtt:284]: MQTT Connected!
[13:30:31]
[13:30:31]User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

[13:30:31]Unhandled C++ exception: OOM
[13:30:31]
[13:30:31]>>>stack>>>
[13:30:31]
[13:30:31]ctx: cont
[13:30:31]sp: 3ffffdb0 end: 3fffffd0 offset: 0010
[13:30:31]3ffffdc0:  00000001 00000020 000001e1 4022d23c  
[13:30:31]3ffffdd0:  00000072 000000f1 3ffffea0 402470af  
[13:30:31]3ffffde0:  000001e0 00000020 3fffa2dc 40101300  
[13:30:31]3ffffdf0:  00000000 000000f0 00000000 00000060  
[13:30:31]3ffffe00:  fffffe04 00000000 3fffbbc0 00000022  
[13:30:31]3ffffe10:  000000f0 000000f1 3ffffea0 402472ac  
[13:30:31]3ffffe20:  fffffe04 00000000 3fffbbd0 00000550  
[13:30:31]3ffffe30:  3fffff48 3fffbb48 3ffffeb8 40208828  
[13:30:31]3ffffe40:  3fffff48 3fffbc40 3ffffeb8 40208856  
[13:30:31]3ffffe50:  3fffff48 3fffbc40 3ffffeb8 40208d4e  
[13:30:31]3ffffe60:  3fffff48 3ffffef0 3ffffea0 40208f0f  
[13:30:31]3ffffe70:  3ffffe78 00000000 3fffbaac 3fffbba0  
[13:30:31]3ffffe80:  3fffbba0 3fffbcb0 3ffffe00 00000000  
[13:30:31]3ffffe90:  3fffbca0 3fffbc30 3ffffe20 00000000  
[13:30:31]3ffffea0:  3fffbd34 000000f0 000000f0 6c635f76  
[13:30:31]3ffffeb0:  223a2261 00776f70 3ffffea0 000000f0  
[13:30:31]3ffffec0:  00000000 00000001 3ffffed0 00000006 <
[13:30:31]3ffffed0:  3ffffed8 00000006 736e6573 00000001  
[13:30:31]3ffffee0:  00000000 3fffff30 3fff4adc 4020cd27  
[13:30:31]3ffffef0:  00000000 69737361 6e617473 3f002f74  
[13:30:31]3fffff00:  3fff8a54 00000012 00000012 40101300  
[13:30:31]3fffff10:  3fff9694 3fffb984 00000004 00000102  
[13:30:31]3fffff20:  3fff4adc 3fff4c38 3fff5eec 4020df7f  
[13:30:31]3fffff30:  3fffba2c 00000055 00000078 00000102  
[13:30:31]3fffff40:  3fff5eec 00000037 3fff5eec 4020ea60  
[13:30:31]3fffff50:  40231438 4020ea34 00000000 6576615f  
[13:30:32]3fffff60:  3fff5eec 00000037 3fff5eec 4020e03a  
[13:30:32]3fffff70:  3fff5dd4 00000036 3fff1d68 4021704d  
[13:30:32]3fffff80:  0000002f 0000002f 3fffa86c 3fff7c44  
[13:30:32]3fffff90:  000000e0 000000b8 0000002e 3fff7cd4  
[13:30:32]3fffffa0:  feefeffe feefeffe feefeffe 3fff2310  
[13:30:32]3fffffb0:  3fffdad0 00000000 3fff22e4 4022d7e8  
[13:30:32]3fffffc0:  feefeffe feefeffe 3fffdab0 40100745  
[13:30:32]<<<stack<<<
[13:30:32]
[13:30:32]last failed alloc call: 402470AF(481)
[13:30:32]
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

[13:30:32]last failed alloc caller: 0x402470af
[13:30:32]
[13:30:32] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[13:30:32]
[13:30:32]load 0x4010f000, len 3424, room 16 
[13:30:32]tail 0
[13:30:32]chksum 0x2e
[13:30:32]load 0x3fff20b8, len 40, room 8 
[13:30:32]tail 0
[13:30:32]chksum 0x2b
[13:30:32]csum 0x2b
[13:30:32]v000a1b50
[13:30:32]~ld
[13:30:32]��n�r��n|��l�l`bbrl�nB�nl`�rl�l��[I][logger:034]: Log initialized
[13:30:32][I][app:029]: Running through setup()...
[13:30:32][W][component:157]: Component wifi set Warning flag: scanning for networks
[13:30:32][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:32][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:32][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:32][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:33][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:33][W][modbus_controller:027]: Modbus device=1 set offline
[13:30:33][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:30:33][W][modbus:114]: Modbus CRC Check failed! C071!=00
```

trianglesis, 2024-09-07:
Trying to add other options under the section `number`, like "# Output voltage".
The board comes back after 2-3 reboots, and fire errors:

```log
Linking .pioenvs/esphome-web-64e0e5/firmware.elf
RAM:   [=====     ]  46.3% (used 37904 bytes from 81920 bytes)
Flash: [=====     ]  49.1% (used 502405 bytes from 1023984 bytes)
Building .pioenvs/esphome-web-64e0e5/firmware.bin
esp8266_copy_factory_bin([".pioenvs/esphome-web-64e0e5/firmware.bin"], [".pioenvs/esphome-web-64e0e5/firmware.elf"])
esp8266_copy_ota_bin([".pioenvs/esphome-web-64e0e5/firmware.bin"], [".pioenvs/esphome-web-64e0e5/firmware.elf"])
========================= [SUCCESS] Took 41.22 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.105
INFO Uploading /data/build/esphome-web-64e0e5/.pioenvs/esphome-web-64e0e5/firmware.bin (506560 bytes)
INFO Compressed to 347137 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 16.03 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.105 using esphome API
INFO Successfully connected to esphome-web-64e0e5 @ 192.168.1.105 in 19.439s
INFO Successful handshake with esphome-web-64e0e5 @ 192.168.1.105 in 0.040s
[13:42:26][I][app:100]: ESPHome version 2024.8.3 compiled on Sep  7 2024, 13:41:45
[13:42:26][I][app:102]: Project esphome.web version dev
[13:42:26][W][component:170]: Component mqtt cleared Warning flag
WARNING esphome-web-64e0e5 @ 192.168.1.105: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esphome-web-64e0e5 @ 192.168.1.105
WARNING Disconnected from API
INFO Successfully connected to esphome-web-64e0e5 @ 192.168.1.105 in 0.153s
INFO Successful handshake with esphome-web-64e0e5 @ 192.168.1.105 in 0.263s
[13:43:06][W][component:170]: Component mqtt cleared Warning flag
[13:43:07][W][safe_mode:028]: Last reset occurred too quickly; safe mode will be invoked in 8 restarts
[13:43:10][E][json:041]: Could not allocate memory for JSON document! Requested 264 bytes, largest free heap block: 264 bytes
[13:43:56][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[13:44:09][E][json:041]: Could not allocate memory for JSON document! Requested 264 bytes, largest free heap block: 264 bytes
```

syssi, 2024-09-07:
```
[13:30:31]Unhandled C++ exception: OOM
```

This is an out of memory exception. If you have the chance please use an ESP32 instead of an ESP8266. The ESP32 is more powerful and able to handle all entities + mqtt.

trianglesis, 2024-09-07:
> ```
> [13:30:31]Unhandled C++ exception: OOM
> ```
> 
> This is an out of memory exception. If you have the chance please use an ESP32 instead of an ESP8266. The ESP32 is more powerful and able to handle all entities + mqtt.

Thanks. It seems like that, but I needed clarification since this is my first time using such tech.

syssi, 2024-09-07:
Nevertheless you did a great job figure this out on your own! :-) Can we close this issue?

trianglesis, 2024-09-07:
> Nevertheless you did a great job figure this out on your own! :-) Can we close this issue?

This is not an issue after all!
Thanks, please close it now.

closed 2024-09-07 by syssi