syssi/esphome-smg-ii #72: Easun SMW 11KW

Alexandros89, 2024-09-20:
hi, 
i'm using a k868 a6 board that have the rs232 port, i have connected to the inverter with his original cable.
the firmware that i'm using is esphome with this configuration:
uart:
  - id: uart_0
    baud_rate: 9600
    debug:
    tx_pin: GPIO16
    rx_pin: GPIO17

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: smg0
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

and my output is this:
[11:17:05][D][uart_debug:114]: >>> 01:03:01:4B:00:07:75:E2
[11:17:05][D][uart_debug:114]: >>> 01:03:01:4B:00:07:75:E2
[11:17:05][D][uart_debug:114]: >>> 01:03:01:4B:00:07:75:E2
[11:17:05][D][uart_debug:114]: >>> 01:03:01:4B:00:07:75:E2
[11:17:05][D][modbus_controller:040]: Modbus command to device=1 register=0x14B countdown=0 no response received - removed from send queue
[11:17:05][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[11:17:06][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[11:17:06][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[11:17:06][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[11:17:06][D][uart_debug:114]: >>> 01:03:01:96:00:01:65:DA
[11:17:06][D][modbus_controller:040]: Modbus command to device=1 register=0x196 countdown=0 no response received - removed from send queue
[11:17:07][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[11:17:07][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[11:17:07][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[11:17:07][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[11:17:07][D][uart_debug:114]: >>> 01:03:00:64:00:02:85:D4
[11:17:08][D][modbus_controller:040]: Modbus command to device=1 register=0x64 countdown=0 no response received - removed from send queue
[11:17:08][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[11:17:08][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[11:17:08][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[11:17:08][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[11:17:09][D][uart_debug:114]: >>> 01:03:00:6C:00:02:04:16
[11:17:09][D][modbus_controller:040]: Modbus command to device=1 register=0x6C countdown=0 no response received - removed from send queue
[11:17:09][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[11:17:09][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=201 count=17
[11:17:09][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=219 count=2
[11:17:09][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=223 count=5
[11:17:09][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=229 count=1
[11:17:09][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=232 count=3
[11:17:09][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=300 count=4
[11:17:09][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=420 count=1
[11:17:09][W][modbus_controller:183]: Duplicate modbus command found: type=0x3 address=643 count=1
[11:17:09][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[11:17:09][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[11:17:09][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[11:17:10][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[11:17:10][D][modbus_controller:040]: Modbus command to device=1 register=0xC9 countdown=0 no response received - removed from send queue
[11:17:10][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[11:17:10][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[11:17:10][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[11:17:11][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[11:17:11][D][uart_debug:114]: >>> 01:03:00:DB:00:02:B4:30
[11:17:11][D][modbus_controller:040]: Modbus command to device=1 register=0xDB countdown=0 no response received - removed from send queue
[11:17:11][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[11:17:11][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[11:17:11][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[11:17:12][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[11:17:12][D][uart_debug:114]: >>> 01:03:00:DF:00:05:B4:33
[11:17:12][D][modbus_controller:040]: Modbus command to device=1 register=0xDF countdown=0 no response received - removed from send queue
[11:17:12][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[11:17:12][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[11:17:13][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[11:17:13][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[11:17:13][D][uart_debug:114]: >>> 01:03:00:E5:00:01:95:FD
[11:17:13][D][modbus_controller:040]: Modbus command to device=1 register=0xE5 countdown=0 no response received - removed from send queue
[11:17:13][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[11:17:13][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[11:17:14][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[11:17:14][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[11:17:14][D][uart_debug:114]: >>> 01:03:00:E8:00:03:85:FF
[11:17:14][D][modbus_controller:040]: Modbus command to device=1 register=0xE8 countdown=0 no response received - removed from send queue
[11:17:14][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[11:17:14][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[11:17:15][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[11:17:15][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[11:17:15][D][uart_debug:114]: >>> 01:03:01:2C:00:04:84:3C
[11:17:15][D][modbus_controller:040]: Modbus command to device=1 register=0x12C countdown=0 no response received - removed from send queue
[11:17:15][D][uart_debug:114]: >>> 01:03:01:A4:00:01:C4:15
[11:17:16][D][uart_debug:114]: >>> 01:03:01:A4:00:01:C4:15
[11:17:16][D][uart_debug:114]: >>> 01:03:01:A4:00:01:C4:15
[11:17:16][D][uart_debug:114]: >>> 01:03:01:A4:00:01:C4:15
[11:17:16][D][uart_debug:114]: >>> 01:03:01:A4:00:01:C4:15
[11:17:16][D][modbus_controller:040]: Modbus command to device=1 register=0x1A4 countdown=0 no response received - removed from send queue
[11:17:16][D][uart_debug:114]: >>> 01:03:02:83:00:01:74:5A
[11:17:17][D][uart_debug:114]: >>> 01:03:02:83:00:01:74:5A
[11:17:17][D][uart_debug:114]: >>> 01:03:02:83:00:01:74:5A
[11:17:17][D][uart_debug:114]: >>> 01:03:02:83:00:01:74:5A
[11:17:17][D][uart_debug:114]: >>> 01:03:02:83:00:01:74:5A

but on home assistant i don't read anithyng.

syssi, 2024-09-20:
This prefix (`>>>`) indicates outgoing traffic / requests. Your log doesn't show any responses (`<<<`). In other words: Your inverter does respond. Please double check your wiring and RS232 converter.

syssi, 2024-09-20:
Do you know anything about the supported protocols of your inverter?

Alexandros89, 2024-09-20:
[image]

I think that the cable shouldn't have problem.

I have asked the seller if he send me the modbus protocol

syssi, 2024-09-20:
Why is a gender changer required if you use the original cable? Could you provide photos of the complete setup (Inverter<->RS232 converter<->ESP).

Alexandros89, 2024-09-20:
This is the board
[image]

This is the original cable
[image]

[image]

The board has natively rs232 port

syssi, 2024-09-20:
I've checked the description of the  k868 a6 board:

[image]

[image]

Using `GPIO16` and `GPIO17` looks good but I guess it doesn't work because RX & TX are swapped because of the wrong gender of the connector.

Alexandros89, 2024-09-20:
In fact I use this board for this reason. I have checked before to use. I have documented.
I have try also to exchange on esphome the gpio but the result is the same.
I will check the cable with the multimeter to understand the pin.

syssi, 2024-09-20:
Swapping the GPIOs at the YAML won't work because the GPIOs are attached to the MAX3232 which has dedicated input and output pins. Do you remember the 0-modem cable from the 90s? ;-) This adapter did cross the RX/TX lines of a serial cable. Connecting two PCs using a RS232 cable + a gender changer won't work too.

Alexandros89, 2024-09-20:
In fact initially I was looking to use a old serial cable but some days ago I wasted all old cable 🙂‍↕️ 
I can change the rj45 cable and crimp newer exchanging the cable 1 and 2?

syssi, 2024-09-20:
> I can change the rj45 cable and crimp newer exchanging the cable 1 and 2?

Please measure the lines first but I guess this would solve the issue.

Alexandros89, 2024-09-20:
I  have tried this
[image]

[image]

But I don't see anything

syssi, 2024-09-20:
Could you try the PIP protocol as next step? F.e. using this YAML: https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml

Do you have another ESP + RS232 converter so we could verify the setup is working in general?

Alexandros89, 2024-09-20:
[image]

[image]

[image]

Seems that the inverter doesn't respond

syssi, 2024-09-20:
Did you try to swap the GPIOs another time?

syssi, 2024-09-20:
Could you bridge the TX to the RX pin of the DSUB connector and test using this loopback? Do you see the outgoing traffic as incoming traffic?

Alexandros89, 2024-09-20:
Yes, I have tried. 
Now I tried also to connect the rs485 of the inverter to the board but the result is the same

Alexandros89, 2024-09-20:
[image]

This is the log with the bridge and the dummy receiver and log both directions

Alexandros89, 2024-09-21:
I have done the bridge in this way
[image]

syssi, 2024-09-21:
There is still no incoming traffic. Something is wrong with your setup. Did you try to swap the GPIOs assigned to the UART another time?

syssi, 2024-09-21:
Feel free to ask more questions if something isn't clear yet.

Alexandros89, 2024-09-21:
I try to swap this morning without any good result.
This evening searching at home I find 3 serial cable. Tomorrow I will try :
Check the serial communication from the inverter, I will connect to the pc and open a serial terminal and look if it send any value
Check the rs232 of the board with an original cable. I think that the wires inside the db9 pinout doesn't make good contact

syssi, 2024-09-22:
I try to make some findings explicit to provide some guidance. Your ESP32+RS232 board doesn't work as expected at the moment. If you bridge TX and RX we should receive an echo (`<<<`) of every outgoing frame (`>>>`). Some ideas what needs to be checked:

1. Please test the continuity between the DB9 pins (`TXD` (p2), `RXD` (p3), `GND?  (p5)) and the MAX3232 (`R1OUT` (p12)
`T1IN` (p11)).
2. Test the continuity of the logic level signals of the MAX3232 (`R1IN` (p13), `T1OUT` (p14)) and `GPIO16` + `GPIO17`.
3. Power the k868 a6 board and measure the voltage between the DB9 pin2 (`TXD`) and pin5 (`GND`).

Alexandros89, 2024-09-22:
[image]

I have check with the multimeter.
If on the 3232 chip I bridge 13 and 14 pin I see on the log some code received.
If I bridge the 11 an 12, I don't see anything in the log

syssi, 2024-09-22:
[image]

Could you show your log? Your observation seems not logic.

Alexandros89, 2024-09-22:
With this bridge

[image]
 I have this log

[image]

----------------
 With this bridge 

[image]

I have this log

[image]

syssi, 2024-09-22:
Okay. This makes more sense. In other words: If we bridge `GPIO16` and `GPIO17` directly (at logical level = 3.3V) the loopback works fine. The `uart` of your ESP32 works in general. The MAX3232 transforms the signal vom logical level 3.3V to the RS232 voltage level of -12/+12V. If we bridge the RS232 TXD/RXD pins the loopback doesn't work.

As next step we have to make sure your MAX3232 is properly powered. Could you measure the voltage between pin 15 and 16? And could you measure the voltage between the RS232 TXD/RXD pins and GND (p13<->GND, p14<->GND)?

Alexandros89, 2024-09-22:
[image]

These are the voltages

Alexandros89, 2024-09-22:
I have found an old rs232 module that I have used when I was very young. 
[image]

I connect it and this is the log
[image]

It works.
Then the problem is the rs232 module of the board

Alexandros89, 2024-09-22:
Connecting this module
 [image]
 I have this output

[image]

syssi, 2024-09-22:
This garbage happens most of the time if GND isn't connected properly.

Alexandros89, 2024-09-22:
Ok now I check better

syssi, 2024-09-22:
This looks like garbage only. Please share plaintext logs instead of screenshots!

What are you doing exactly at the moment?

closed 2024-12-26 by syssi

Alexandros89, 2025-01-01:
Hi,
I haven't had any times previously, for this reason I haven't put any news.
But during this time I ordered the new device.
An esp32 with Lan the wt32, I connect a rs232 converter and yesterday I started to read some information from the inverter.
Now the question is that in the log I see the value but on home assistant no.
I add the log

INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/inverter1.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.1.71 using esphome API
INFO Successfully connected to inverter1 @ 192.168.1.71 in 0.262s
INFO Successful handshake with inverter1 @ 192.168.1.71 in 0.082s
[02:00:41][I][app:100]: ESPHome version 2024.12.2 compiled on Jan  1 2025, 01:54:19
[02:00:41][C][wifi:600]: WiFi:
[02:00:41][C][wifi:428]:   Local MAC: 88:13:BF:C8:02:D8
[02:00:41][C][wifi:433]:   SSID: [redacted]
[02:00:41][C][wifi:436]:   IP Address: 192.168.1.71
[02:00:41][C][wifi:440]:   BSSID: [redacted]
[02:00:41][C][wifi:441]:   Hostname: 'inverter1'
[02:00:41][C][wifi:443]:   Signal strength: -56 dB ▂▄▆█
[02:00:41][C][wifi:447]:   Channel: 9
[02:00:41][C][wifi:448]:   Subnet: 255.255.255.0
[02:00:41][C][wifi:449]:   Gateway: 192.168.1.1
[02:00:41][C][wifi:450]:   DNS1: 192.168.1.1
[02:00:41][C][wifi:451]:   DNS2: 0.0.0.0
[02:00:41][C][logger:185]: Logger:
[02:00:41][C][logger:186]:   Level: DEBUG
[02:00:41][C][logger:188]:   Log Baud Rate: 115200
[02:00:42][C][logger:189]:   Hardware UART: UART0
[02:00:42][C][uart.arduino_esp32:151]: UART Bus 1:
[02:00:42][C][uart.arduino_esp32:152]:   TX Pin: GPIO17
[02:00:42][C][uart.arduino_esp32:153]:   RX Pin: GPIO5
[02:00:42][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[02:00:42][C][uart.arduino_esp32:157]:   Baud Rate: 2400 baud
[02:00:42][C][uart.arduino_esp32:158]:   Data Bits: 8
[02:00:42][C][uart.arduino_esp32:159]:   Parity: NONE
[02:00:42][C][uart.arduino_esp32:160]:   Stop bits: 1
[02:00:42][D][uart_debug:158]: >>> "QMODI\xC1\r"
[02:00:42][D][pipsolar:842]: Sending polling command : QMOD with length 4
[02:00:42][C][pipsolar:864]: Pipsolar:
[02:00:42][C][pipsolar:865]: used commands:
[02:00:42][C][pipsolar:868]: QPIRI
[02:00:42][C][pipsolar:868]: QPIGS
[02:00:42][C][pipsolar:868]: QMOD
[02:00:42][C][pipsolar:868]: QFLAG
[02:00:42][C][pipsolar.switch:068]: Pipsolar Switch 'smw-ii output_source_priority_utility'
[02:00:42][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[02:00:42][C][pipsolar.switch:091]:   Restore Mode: always OFF
[02:00:42][D][uart_debug:158]: <<< "(B\xE7\xC9\r"
[02:00:42][C][pipsolar.switch:068]: Pipsolar Switch 'smw-ii output_source_priority_solar'
[02:00:42][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[02:00:42][C][pipsolar.switch:091]:   Restore Mode: always OFF
[02:00:42][D][pipsolar:776]: checking crc on incoming message
[02:00:42][D][pipsolar:779]: CRC OK
[02:00:42][C][pipsolar.switch:068]: Pipsolar Switch 'smw-ii output_source_priority_battery'
[02:00:42][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[02:00:42][C][pipsolar.switch:091]:   Restore Mode: always OFF
[02:00:42][D][pipsolar:475]: Decode QMOD
[02:00:42][D][text_sensor:064]: 'smw-ii last_qmod': Sending state '(B'
[02:00:42][C][pipsolar.switch:068]: Pipsolar Switch 'smw-ii input_voltage_range'
[02:00:42][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[02:00:42][C][pipsolar.switch:091]:   Restore Mode: always OFF
[02:00:42][D][text_sensor:064]: 'smw-ii device_mode': Sending state 'Battery mode'
[02:00:42][C][pipsolar.switch:068]: Pipsolar Switch 'smw-ii pv_ok_condition_for_parallel'
[02:00:42][C][pipsolar.switch:070]:   Icon: 'mdi:power'
[02:00:42][C][pipsolar.switch:091]:   Restore Mode: always OFF
[02:00:42][C][pipsolar.switch:068]: Pipsolar Switch 'smw-ii pv_power_balance'
[... 117 lines cut ...]
[02:01:03][D][uart_debug:158]: <<< "(EbxzDadjklnuvyI7\r"
[02:01:03][D][pipsolar:776]: checking crc on incoming message
[02:01:03][D][pipsolar:779]: CRC OK
[02:01:03][D][pipsolar:483]: Decode QFLAG
[02:01:03][D][text_sensor:064]: 'smw-ii last_qflag': Sending state '(EbxzDadjklnuvy'
[02:01:04][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[02:01:04][D][pipsolar:842]: Sending polling command : QPIRI with length 5
[02:01:05][D][uart_debug:158]: <<< "(230.0 47.8 230.0 50.0 47.8 11000 11000 48.0 47.0 46.4 56.9 56.9 3 040 150 1 2 3 9 01 0 0 52.0 0 1 480 0 000\xE1S\r"
[02:01:05][D][pipsolar:776]: checking crc on incoming message
[02:01:05][D][pipsolar:786]: CRC NOK expected: E0 D0 but got: 0 0
[02:01:05][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[02:01:05][D][pipsolar:842]: Sending polling command : QPIGS with length 5
[02:01:06][D][uart_debug:158]: <<< "(226.9 50.0 230.2 49.9 01150 01063 011 351 50.40 000 047 0022 00.0 000.0 00.00 00024 00010000 00 00 00000 010 0 00 0000r\x02\r"
[02:01:10][D][pipsolar:759]: timeout command to poll: QPIGS
[02:01:10][D][uart_debug:158]: >>> "QMODI\xC1\r"
[02:01:10][D][pipsolar:842]: Sending polling command : QMOD with length 4
[02:01:10][D][uart_debug:158]: <<< "(B\xE7\xC9\r"
[02:01:10][D][pipsolar:776]: checking crc on incoming message
[02:01:10][D][pipsolar:779]: CRC OK
[02:01:10][D][pipsolar:475]: Decode QMOD
[02:01:10][D][text_sensor:064]: 'smw-ii last_qmod': Sending state '(B'
[02:01:10][D][text_sensor:064]: 'smw-ii device_mode': Sending state 'Battery mode'
[02:01:11][D][uart_debug:158]: >>> "QFLAG\x98t\r"

[image]