syssi/esphome-smg-ii #73: Loop upgrade

chernienkoii, 2024-09-24:
Hello!
When i use esp8266 always get a problem with loop upgrade
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 2.0.0
|-- ArduinoJson @ 6.18.5
RAM:   [======    ]  58.5% (used 47960 bytes from 81920 bytes)
Flash: [======    ]  57.2% (used 597193 bytes from 1044464 bytes)
========================= [SUCCESS] Took 1.02 seconds =========================

And after that start upgrade:

```
INFO Successfully compiled program.
Found multiple options for uploading, please choose one:
  [1] /dev/ttyUSB0 (USB Serial)
  [2] Over The Air (smg-ii.local)
(number): 1
esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a4:cf:12:dd:1a:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x00092fff...
Compressed 601344 bytes to 385695...
Wrote 601344 bytes (385695 compressed) at 0x00000000 in 8.9 seconds (effective 542.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[16:42:53]{dd\x9c\x9f|\x84l\xe0|
                                \xc4
                                    l\xecc|\x87\x82\xe4x9br\x93c\x84c\x8c\xf3'n\x9flgg\x9c\xe3\xec
                  cx\x84\x8f$sd{l8\xf3o\xe0
                                           \x83
                                               l
                                                \x84\x9c

                                                        #o\xe2|\xe4\x8f\x98
                                                                           #\x84\xfbog\xefdćl \x98o
                   d`gs\x8e\x9b\xdbg
                                    cl\x9c$p\xfbo\xe0{\x8cܜ\x9c\x80\x8cbg\xe3|\x8f
  c\x84\xfboo\xef
                 d`\x98no
                         l`o;\x87\x93\x93o

                                          c\x84\x9b`s\x93\x9bo

                                                              c\x84\x9b`\x9cl\x8c
 \x9c$`\xfc\x83n\x9cd\x8c\xe3\xecx82g\xec\x93{\x83\xfbg|\xe4
                                                            d\xc4
                                                                 l`c\x9b
                                                                        x9e|{\x9bl\xdc'\xe0
           \x83o\xe2
                    l`\x8c\xe3{\x93l\x8cd\x8c

[... 449 lines cut ...]
WARNING Memory allocation of 268 bytes failed at 40218CAB
WARNING Decoded 0x40218cab: esp8266::MDNSImplementation::MDNSResponder::_sendHostProbe()
[16:43:09]
[16:43:09]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[16:43:09]
[16:43:09]last failed alloc caller: 0x40218cab
[16:43:09]
[16:43:09] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[16:43:09]
[16:43:09]load 0x4010f000, len 3424, room 16 
[16:43:09]tail 0
[16:43:09]chksum 0x2e
[16:43:09]load 0x3fff20b8, len 40, room 8 
[16:43:09]tail 0
[16:43:09]chksum 0x2b
[16:43:09]csum 0x2b
[16:43:09]v00092d00
[16:43:09]~ld
[16:43:09]\x84\xe3\xe4x82g\xec\x93{\x83\xf2g|\xec$
                                                  \x8c
                                                      d`b\x93\x9f|r\x93d\x9co\xe0\x83'\xe3l`\xc4\xe3{\x92l\x8clx84

                                  \xfe[I][logger:034]: Log initialized

```

syssi, 2024-09-24:
This (`[16:43:09]Unhandled C++ exception: OOM`) is an out of memory exception. Please reduce your YAML or switch to an ESP32.

Haso-M, 2024-11-03:
Works if you use API instead of MQTT, but in my case updating values from the inveter was not reliable. Tested on Wifi Plug Pro-05 that came with Easun SMG-III 6200, with few modifications to reset the restart timer ciurcut on this devices.

closed 2024-12-26 by syssi