syssi/esphome-smg-ii #75: Inverter SMG II Easun problem esphome 9.2

spikeshun, 2024-10-09:
Hi!

I updated the esphome version and since then i am not getting any data. 
What can i do?

syssi, 2024-10-09:
If you downgrade to the previous ESPHome version does it work again?

spikeshun, 2024-10-09:
I can't downgrade:(

spikeshun, 2024-10-09:
its my config: 

substitutions:
  name: smg-ii
  device_description: "Monitor and control a ISolar/EASUN SMG II inverter via RS232"
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  platform: ESP32
  board: lolin32
  comment: ${device_description}
  project:
    name: "syssi.esphome-smg-ii"
    version: 1.0.0

wifi:
  ssid: !secret SSID
  password: !secret wifi_password
  manual_ip:
    static_ip: xxx.xxx.xxx.xxx
    gateway: xxx.xxx.xxx.xxx
    subnet: 255.255.255.0
  power_save_mode: none
  
ota:
  platform: esphome
  password: "xxxx"

logger:
  level: DEBUG

api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: smg0
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

time:
  - platform: sntp

syssi, 2024-10-09:
Do you have a linux host / Debian based machine at your network?

ESPHome can be installed pretty easily on the command line:

```
$ sudo apt-get install pipx
$ pipx install esphome==2024.6.0
Installing to existing venv 'esphome'
  installed package esphome 2024.6.0, installed using Python 3.11.2
  These apps are now globally available
    - esphome
done! ✨ 🌟 ✨
$ esphome version
Version: 2024.6.0

$ esphome run yourconfig.yaml
```

spikeshun, 2024-10-09:
I understand this, but if I go back with other integrations I may have problems.

syssi, 2024-10-09:
It's just to make sure the ESPHome version is the root cause.

syssi, 2024-10-09:
I guess it won't work with the old ESPHome version too and we have to reboot your inverter or inspect the RS232 converter.

spikeshun, 2024-10-09:
But I did nothing but update this one device.

spikeshun, 2024-10-09:
It has worked perfectly so far, never had a problem with it. I pressed update yesterday and it went away. Unfortunately, I don't know what version it was before, but I think it was something like 7 or 8. I don't think there would be a problem with the RSR232 converter.

syssi, 2024-10-09:
We have to identify the root cause. The best start is building the project with different ESPHome versions and compare the logs.

spikeshun, 2024-10-10:
I tried the following modification, but it still doesn't work :(

esphome:
  name: ${name}
#  platform: ESP32
#  board: lolin32
  comment: ${device_description}
  project:
    name: "syssi.esphome-smg-ii"
    version: 1.2.0

esp32:
  board: esp32dev
  framework:
    type: arduino

syssi, 2024-10-13:
Do you have another RS232 converter and ESP for doing some tests?

syssi, 2024-10-22:
Friendly reminder.

closed 2024-12-26 by syssi