syssi/esphome-smg-ii #76: EASUN SMH II 4.2K

avenseElectro, 2024-10-22:
Hello, can you get this inverter to work?
For now with your esphome I still haven't managed to do it.
Below I show you the esphome log and the watchpower debug

```Esphome log
11:42:40 | [D] | [uart_debug:114] | >>> 01:03:00:E8:00:03:85:FF
11:42:40 | [W] | [modbus:114] | Modbus CRC Check failed! C071!=00
11:42:40 | [D] | [uart_debug:114] | >>> 01:03:00:E8:00:03:85:FF
11:42:40 | [D] | [uart_debug:114] | >>> 01:03:00:E8:00:03:85:FF
11:42:40 | [D] | [uart_debug:114] | <<< 00:00:00:00:00:00:00
11:42:40 | [D] | [uart_debug:114] | >>> 01:03:00:E8:00:03:85:FF
11:42:40 | [W] | [modbus:114] | Modbus CRC Check failed! C071!=00
11:42:41 | [D] | [modbus_controller:037] | Modbus command to device=1 register=0xE8 no response received - removed from send queue
11:42:41 | [D] | [uart_debug:114] | >>> 01:03:00:64:00:02:85:D4
11:42:41 | [W] | [modbus:114] | Modbus CRC Check failed! C071!=00
11:42:41 | [D] | [uart_debug:114] | <<< 00:00:00:00:00:00:00:00
11:42:41 | [D] | [uart_debug:114] | >>> 01:03:00:64:00:02:85:D4
11:42:41 | [W] | [modbus:114] | Modbus CRC Check failed! C071!=00
11:42:42 | [D] | [uart_debug:114] | <<< 00:00:00:00:00:00:00:00
11:42:42 | [D] | [uart_debug:114] | >>> 01:03:00:64:00:02:85:D4
11:42:42 | [W] | [modbus:114] | Modbus CRC Check failed! C071!=00
11:42:42 | [D] | [uart_debug:114] | <<< 00:00:00:00:00:00:00:00
11:42:42 | [D] | [uart_debug:114] | >>> 01:03:00:64:00:02:85:D4
11:42:45 | [W] | [modbus:114] | Modbus CRC Check failed! C071!=00
11:42:45 | [D] | [uart_debug:114] | <<< 00:00:00:00:00:00:00:00
11:42:45 | [D] | [uart_debug:114] |  
```
[debug.zip](https://github.com/user-attachments/files/17474862/debug.zip)

syssi, 2024-10-22:
It looks like your wiring isn't fine or your RS232 is faulty. Could you provide a photo?

avenseElectro, 2024-10-22:
> It looks like your wiring isn't fine or your RS232 is faulty. Could you provide a photo?

of course.

[image]
[image]
[image]
[image]
[image]

syssi, 2024-10-22:
Could you provide another photo of your RJ45 plug?

avenseElectro, 2024-10-22:
> Could you provide another photo of your RJ45 plug?

There ir os
[image]
[image]
[image]
[image]

syssi, 2024-10-22:
Okay. You setup looks good in general! Could you provide your YAML configuration as last resort? :-)

syssi, 2024-10-22:
Could you swap the RX/TX gpios at the YAML and reflash your ESP just for testing?

avenseElectro, 2024-10-23:
> Could you swap the RX/TX gpios at the YAML and reflash your ESP just for testing?

It didn't work....
Did you see my inverter, didn't you see it's in the title
EASUN SMH II 4.2K

syssi, 2024-10-23:
It looks like your inverter responds to PIP commands:

```
head -n3 2024-10-21\ Serial-QPIGS.log 
[2024-10-21 18:18:01](000.0 00.0 230.0 49.9 0023 0014 000 392 26.40 001 100 0043 0001 150.5 00.00 00000 00010110 00 00 00021 110
[2024-10-21 18:18:04](000.0 00.0 229.9 49.9 0022 0002 000 392 26.40 001 100 0043 0001 149.7 00.00 00000 00010110 00 00 00021 110
[2024-10-21 18:18:09](000.0 00.0 230.2 49.9 0023 0004 000 392 26.40 001 100 0043 0001 149.7 00.00 00000 00010110 00 00 00020 110
```

Did you try `esphome-pipsolar` instead?

avenseElectro, 2024-10-23:
> It looks like your inverter responds to PIP commands:
>
> ```
> $ head -n3 2024-10-21\ Serial-QPIGS.log
> [2024-10-21 18:18:01](000.0 00.0 230.0 49.9 0023 0014 000 392 26.40 001 100 0043 0001 150.5 00.00 [...]

Yes, I tested pipsolar before

```
12:33:12 | [D] | [pipsolar:759] | timeout command to poll: QPIGS
-- | -- | -- | --
12:33:13 | [D] | [pipsolar:840] | Sending polling command : QMOD with length 4
12:33:16 | [D] | [pipsolar:759] | timeout command to poll: QMOD
12:33:17 | [D] | [pipsolar:840] | Sending polling command : QFLAG with length 5
12:33:21 | [D] | [pipsolar:759] | timeout command to poll: QFLAG
12:33:22 | [D] | [pipsolar:840] | Sending polling command : QPIRI with length 5
12:33:27 | [D] | [pipsolar:759] | timeout command to poll: QPIRI
12:33:27 | [D] | [pipsolar:840] | Sending polling command : QPIGS with length 5
12:33:32 | [D] | [pipsolar:759] | timeout command to poll: QPIGS
12:33:32 | [D] | [pipsolar:840] | Sending polling command : QMOD with length 4
12:33:37 | [D] | [pipsolar:759] | timeout command to poll: QMOD
12:33:37 | [D] | [pipsolar:840] | Sending polling command : QFLAG with length 5
12:33:42 | [D] | [pipsolar:759] | timeout command to poll: QFLAG
12:33:42 | [D] | [pipsolar:840] | Sending polling command : QPIRI with length 5
12:33:47 | [D] | [pipsolar:759] | timeout command to poll: QPIRI
12:33:47 | [D] | [pipsolar:840] | Sending polling command : QPIGS with length 5
12:33:52 | [D] | [pipsolar:759] | timeout command to poll: QPIGS
12:33:52 | [D] | [pipsolar:840] | Sending polling command : QMOD with length 4
```

syssi, 2024-10-23:
Please give it another try. Do you have a second ESP and RS232 converter? Could we assemble a test setup which looks like this:

```
ESP (a) <--> MAX3232 (a) <-----------> MAX3232 (b) <--> ESP (b)
```

We can use this setup to verify all components are wired properly.

avenseElectro, 2024-10-23:
> Please give it another try. Do you have a second ESP and RS232 converter? Could we assemble a test setup which looks like this:
> 
> ```
> ESP (a) <--> MAX3232 (a) <-----------> MAX3232 (b) <--> ESP (b)
> ```
> 
> We can use this setup to verify all components are wired properly.

I have a second inverter with pip30 protocol and as soon as I put it in the inverter it starts working, so I believe that everything is ok with the components and their connection.
I have already sent an email to EASUN asking about the communication protocol...

syssi, 2024-10-23:
If you use watchpower do you use a RS232 connection too? Do you use a special cable?

avenseElectro, 2024-10-23:
> If you use watchpower do you use a RS232 connection too? Do you use a special cable?

I use the original cable that came with the equipment to connect the wifi, then I connect the usb/232 to the pc.

syssi, 2024-10-23:
Did you use USB or RS232?

avenseElectro, 2024-10-23:
> Did you use USB or RS232?

My pc doesn't have rs232 port so I use a rs232/usb converter to connect to the pc

syssi, 2024-10-23:
But you are using the RS232 port of the inverter? Could you provide a photo of the OEM RS232 and the RS232-to-USB converter? May be the pin out of the RJ45 is different and we can identify the difference by measuring the lines.

avenseElectro, 2024-10-23:
> But you are using the RS232 port of the inverter? Could you provide a photo of the OEM RS232 a
[image]
[image]
nd the RS232-to-USB converter? May be the pin out of the RJ45 plug is different and we can identify the difference by measuring the lines.
[image]

[image]
[image]

syssi, 2024-10-23:
Good job! Do you have a multimeter? Could you measure the continuity between the 4 used RJ45 pins (orange, white, black and green) and the pins of the D-SUB connector? To which D-SUB pins are the different colors attached?

avenseElectro, 2024-10-23:
> Good job! Do you have a multimeter? Could you measure the continuity between the 4 used RJ45 pins (orange, white, black and green) and the pins of the D-SUB connector? To which D-SUB pins are the different colors attached?

I couldn't see the continuity of the cable. None "whisked".
Do you have any suggestions?

syssi, 2024-10-23:
What did you do exactly? It looks like there is a misunderstanding.

avenseElectro, 2024-10-23:
> What did you do exactly? It looks like there is a misunderstanding.

The fourth wire is the 12v wire for the wifi dongle
That's why this cable has 4 wires.

syssi, 2024-10-23:
You are right. Last idea: Could you check the used baudrate at watchpower?

avenseElectro, 2024-10-23:
> You are right. Last idea: Could you check the used baudrate at watchpower?

[image]

syssi, 2024-10-23:
Could you try the `esphome-pipsolar` project another time and increase the baudrate from 2400 to 19200 baud?

avenseElectro, 2024-10-24:
> Could you try the `esphome-pipsolar` project another time and increase the baudrate from 2400 to 19200 baud?

Unfortunately it still doesn't work....

[image]
[image]

carmelo42, 2024-11-23:
> > What did you do exactly? It looks like there is a misunderstanding.
> 
> The fourth wire is the 12v wire for the wifi dongle That's why this cable has 4 wires.

Hi, I have a SMH II 3.2K which fries my usb to serial adapter. I think it's because of the 12v provided on the DB9 port to power the wifi adapter.
Can you point me the pinout of the original RJ45 cable ? So I'll know which cable provides 12V and I'll be able to cut it off : 

[image]

Will the serial adapter be usable without the 12v ?

syssi, 2025-07-05:
I will close this issue because I'm out of ideas. If you want to give it another try feel free to re-open the issue again.

closed 2025-07-05 by syssi

syssi, 2025-07-05:
This is a similar device (using the pipsolar protocol): https://github.com/syssi/esphome-smg-ii/issues/80