syssi/esphome-smg-ii #79: Connecting the inverter through USB

felipecrs, 2024-11-12:
Hi,

I own a SRNE HF2430U60-100 and I'm eager to try this custom component, but I'm waiting for my RS485-TTL 
and RJ45 to pins adapters to arrive.

So far, I have been pulling data from my inverter through this project: https://github.com/HotNoob/PythonProtocolGateway, which works by connecting the inverter USB to my Home Assistant PC. It exposes and communicates through the same modbus protocol you have been working on #28.

But I own a [ESP32-S3 DevKit C 1](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html), which has two USB-C ports, including USB OTG functionality.

Do you believe it would be possible to make the ESP pull the data from the inverter through USB? I have no expertise whatsoever in this.

This would make the entire process a lot simpler by avoiding the need of an ethernet cable, adapter, and RS485-TTL converter.

syssi, 2024-11-12:
ESPHome doesn't provide USB host support yet. As soon USB host support is available and a USB-ACM devices is registered as `uart` interface this implementation should work out of the box. For me it's too much effort to do the job to avoid a cheap converter board. ;-)

felipecrs, 2024-11-12:
It makes total sense. Do you believe this issue in ESPHome is what I should be looking for?

- https://github.com/esphome/feature-requests/issues/2924

On a side note, I found a project that reads from USB like I want:

https://github.com/ludoux/esp32-nut-server-usbhid

But it's not ESPHome anyhow.

syssi, 2024-11-13:
The mentioned ESPHome PR is about USB client mode. It turns the ESP into an USB client and advertises an serial interface 

You want to enable the USB host mode (OTG?) and be able to enumerate your inverter as `uart`.

syssi, 2024-11-13:
The second project used the ESP host mode but expects an USB-HID (human interface device). If you plug your inverter into a PC it's registered as additional serial port (USB-ACM), correct?

felipecrs, 2024-11-13:
> If you plug your inverter into a PC it's registered as additional serial port (USB-ACM), correct?

Yes, that's right.

syssi, 2024-11-13:
I guess this is what you are looking for: https://components.espressif.com/components/espressif/usb_host_cdc_acm

syssi, 2024-11-13:
A bit off-topic: Could you double check your inverter is equipped with a RS485 interface/port so the RS485 converter is the correct choice? I would like to avoid you have to order a RS232 converter (MAX3232) at some point.

felipecrs, 2024-11-13:
Yes, my inverter has a RJ45 port imprinted as _Wi-Fi RS485_. I intend to follow the schematics of this post:

https://community.home-assistant.io/t/srne-inverter-integration-help/609730

[image]

To make the connection to my device. I wonder why R7 needs to be cut, but I'll probably just do anyway.

felipecrs, 2024-11-13:
@syssi, thank you very much for the answers you provided. I have filled a feature request in ESPHome:

- https://github.com/esphome/feature-requests/issues/2944

And since there's nothing we can here without that, I'm closing this issue.

closed 2025-07-25 by syssi