syssi/esphome-smg-ii #81: Complie fails when using Time, total_daily_energy, Switch blocks

AlbertWeterings, 2024-12-01:
I'm trying to build the project for 2 SMG-II is parallel connected to 1 ESP32
Does anyone have a sollution for the problems below?

When I add in the config:
time:
  - platform: sntp
The errors.. are thrown
undefined reference to `vtable for esphome::sntp::SNTPComponent'
undefined reference to `esphome::time::RealTimeClock::RealTimeClock()'

The part "total_daily_energy" is needing Time so I expect that will be solved after the time issue is solved?

When I add Switches in the config.
The errors.. are thrown together with a bunch more
undefined reference to `esphome::switch_::Switch::turn_on()
undefined reference to `esphome::switch_::Switch::turn_off()
undefined reference to `esphome::switch_::Switch::toggle()
undefined reference to `esphome::switch_::Switch::add_on_state_callback(std::function<void (bool)>&&)

My full config attached with the failing parts uncommented
[esp32-example-advanced-multiple-uarts - Copy.yaml.txt](https://github.com/user-attachments/files/17968858/esp32-example-advanced-multiple-uarts.-.Copy.yaml.txt)

syssi, 2024-12-01:
Did you try a clean build?

AlbertWeterings, 2024-12-01:
What do you mean by a clean build?
If I build it with the attached config there is no issue. The issues only are there when I enable the parts mentioned in the message above.

syssi, 2024-12-01:
The is a `clean` build button at the ESPHome dashboard. If you use the CLI `esphome clean yourconfig.yaml` is your friend.