syssi/esphome-smg-ii #86: Add battery type and SOC limits as W/R holding registers

ogry, 2025-01-14:
As the RTU modbus register to hold battery type (value F2-01 in the manual) is 630, with this possible values:

0: AGM
1: FLD
2: USER
4: li2
6:li4
8:lib

Register 647 holds the % of SOC at with the inverter consider the battery needs to be charged (value F2-05 in the manual, with lithium battery communication) (uint16)

Register 648 holds the % of SOC at with the inverter consider the battery is charged (value F2-06 in the manual, with lithium battery communication)  (uint16)

Register 650 holds the % of SOC to cut-off discharging battery (value F2-08 in the manual, with lithium battery communication) (uint16)

tricky10, 2025-02-15:
These registers does not work for smg III

ogry, 2025-02-15:
> These registers does not work for smg III

May be, this is SMG II, not SMG III

tricky10, 2025-02-16:
@ogry any idea where can I find the modbus specification doc  ?

Zsolt73, 2025-05-11:
> As the RTU modbus register to hold battery type (value F2-01 in the manual) is 630, with this possible values:
>
> 0: AGM 1: FLD 2: USER 4: li2 6:li4 8:lib
>
> Register 647 holds the % of SOC at with the [...]

Hello, Ogry! This registers are for Easun SMG II 11kW inverter. I have some other registers address for thi inverter but not all. Which other registers do you have? Сan you share other registers, which you have? Thank you in advance.

MBRetrofit, 2025-08-20:
As address 630 not work for Easun SMGII 6,2KW, the addresses:

647, 648, 650 need to be mapped too

syssi, 2025-08-22:
@MBRetrofit Did you try to read register 647, 648 and 650 on your SMGII? These registers doesn't fit for your device model, correct?

MBRetrofit, 2025-08-22:
> As address 630 not work for Easun SMGII 6,2KW, the addresses:
> 
> 647, 648, 650 need to be mapped too

630 - 322 = 308

I subtracted 308 from 647, 648, 650 and I began to receive some data...

I will make some more test's

MBRetrofit, 2025-08-22:
```
  # SOC to cut-off discharging battery                            %       Uint  650 1 R/W Value F2-08 (lithium battery communication)
  - platform: modbus_controller
    modbus_controller_id: smg0
    name: "${device0} battery discharge cutoff soc threshold"
    use_write_multiple: true
    address: 341
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 100.0
    step: 1
    unit_of_measurement: "%"
    entity_category: config
    icon: mdi:battery-alert  

````

This OK 100%

I have to find the other 2

syssi, 2025-08-22:
Could you verify the correctness by changing the setting using the on-screen-menu. Does this register change accordingly?

MBRetrofit, 2025-08-28:
Of course,

I don't know if the dependencies has to be established, like:

 Default 50%, 5%~50%
Settable,but the minimum setting
value must be more than the value
of program 45.

and so on,

syssi, 2025-08-28:
This isn't possible (at the YAML configuration and in best case not needed). If the inverter firmware does a good job it should reject invalid values and keep the current/valid value.

MBRetrofit, 2025-08-28:
It does a very good job, but at least the min/max values should be added , what do you think?

If needed, dependencies can be added with Lambda, but I think is ok for now

closed 2025-11-01 by syssi