syssi/esphome-smg-ii #91: Total energy production / last 365 days

RhibRider, 2025-02-28:
Hello, ich can read dayly energy production in kwh, but Total energy production / last 365 days would also be great!
Can this be implemented? Or don't SMGs a counter for that?

syssi, 2025-02-28:
This measurement isn't available. You have to calculate the value on your own.

closed 2025-03-01 by syssi

syssi, 2025-03-01:
You could also add the logic the your YAML configuration:

```
sensor:
  - platform: total_daily_energy
    name: "${name} PV energy today"
    restore: true
    icon: mdi:counter
    power_id: smg0_pv_average_power
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    
  - platform: integration
    name: "${name} PV energy yearly"
    id: pv_energy_yearly
    sensor: smg0_pv_average_power
    time_unit: h
    unit_of_measurement: kWh
    icon: mdi:solar-power-variant
    restore: true
    filters:
      - multiply: 0.001

# Automatic reset at new year
automation:
  - trigger:
      platform: time
      at:
        hours: 0
        minutes: 0
        seconds: 0
    condition:
      condition: time
      # Checks if it's January 1st
      month: 1
      day_of_month: 1
    then:
      - sensor.integration.reset: pv_energy_yearly
```