syssi/esphome-smg-ii #96: Errors

leszekorzel, 2025-03-21:
Hi, 
I have a strange problem. It has happened suddenly after power outage. Before everything was working correctly.
Here is some logs.

[12:11:42][D][modbus_controller:039]: Modbus command to device=1 register=0x64 no response received - removed from send queue
[12:11:42][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[12:11:43][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[12:11:43][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[12:11:43][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=201 count=17
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=232 count=3
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=300 count=4
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=305 count=6
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=313 count=1
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=320 count=2
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=323 count=5
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=329 count=1

syssi, 2025-03-21:
These messages can be ignored:

```
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=201 count=17
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=232 count=3
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=300 count=4
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=305 count=6
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=313 count=1
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=320 count=2
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=323 count=5
[12:11:43][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=329 count=1
```

They are for debugging only (log level `VERY_VERBOSE`).

```
[12:11:42][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[12:11:43][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[12:11:43][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
[12:11:43][D][uart_debug:114]: >>> 01:03:00:C9:00:11:55:F8
```

This is outgoing (`>>>`) traffic. It looks like your inverter doesn't respond (`<<<`). 

This is the implementation complaining about the missing responses of your inverter:

```
[12:11:42][D][modbus_controller:039]: Modbus command to device=1 register=0x64 no response received - removed from send queue
```

leszekorzel, 2025-03-21:
Thank you. Luckily I've checked the TTL converter and it was dead.

closed 2025-03-21 by leszekorzel