syssi/esphome-smg-ii #102: communicate jk-bms and inverter via rs485

test3210-d, 2025-06-07:
I use  esp32c3, adding this part of config work for me, li_4 mode, pylontech is much more difficult to implement. І have different config using bluetooth communication to jk bms but most part of config same

external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s
  - source: github://arpiecodes/esphome-modbus-server@master
    refresh: 0s
    components:
      - modbus_server

uart:
  - id: uart_rs485
    tx_pin: GPIO6
    rx_pin: GPIO5
    baud_rate: 9600
    stop_bits: 1
    parity: NONE
    rx_buffer_size: 512

modbus_server:
  - id: modbuserver
    uart_id: uart_rs485
    address: 1  # slave address
    de_pin: 7  // de + re on rs485 board
    holding_registers:
      - start_address: 19
        number: 17
        on_read: |-
          if (address == 19) return int(107);        // 0x13
          if (address == 20) return int(id(jk485_errors_bitmask).state); 
          if (address == 21) return int(id(jk485_capacity_remaining).state);    // 0x15 SoC
          if (address == 22) return int(id(jk485_total_voltage).state * 100);   // 0x16 Voltage (10mV)
          if (address == 23) return int(id(jk485_current).state * 100);         // 0x17 Current (10mA)
          if (address == 24) return int(id(jk485_temperature_sensor_1).state * 10); // 0x18 Temp (0.1В°C)
          if (address == 25) return int(id(jk485_max_charge).state* 100)-1800;  // 0x19 Max current
          if (address == 26) return int(id(jk485_capacity_remaining).state);    // 0x1A Rem cap
          if (address == 27) return int(id(jk485_actual_battery_capacity).state); // 0x1B Full cap
          if (address == 28) return 0;      // 0x1C Firmware reserved
          if (address == 29) return 10;     // 0x1D ?? Delta V
          if (address == 30) return int(id(jk485_charging_cycles).state);       // 0x1E Cycles
          if (address == 31) return 0;      // 0x1F Reserved
          if (address == 32) return 99;     // 0x20 SoH
          if (address == 33) return int(id(jk485_bat_count).state * id(jk485_volt_rest).state * 100) + int(id(jk485_cv_voltage_offset).state); //  max charge voltage.           
          if (address == 34) return int(id(jk485_errors_bitmask).state);        // 0x22 Warning code
          if (address == 35) return int(id(jk485_max_discharge).state* 100);  // 0x23 Max discharge curent
          return 0;

test3210-d, 2025-06-08:
official modbus don`t work on MAX485 Module , all other function supported. The software library used in the project is able to work with the MAX485 chip, switching reception - transmission via a change in pin level. In all other functions, the library completely identical to the official one. So, if you use another transmitter, you can use the standard library.

test3210-d, 2025-06-09:
I check, work same. On esp32 wire connecton for all ports. If use esp32c3, jk-bms bluetooth communication only, not enough com ports for all.
```
jk_modbus:
  - id: modbus0
    uart_id: uart_1
    rx_timeout: 1150ms

jk_bms:
  - id: bms0
    jk_modbus_id: modbus0
    update_interval: 5s

modbus:
#control modbus 
  - id: modbus1
    uart_id: uart_2
    send_wait_time: 150ms
#battery modbus
  - uart_id: uart_3
    id: modbus_server
    role: server
    flow_control_pin: GPIO27

modbus_controller:
  - id: smg0
    address: 0x01
    modbus_id: modbus1
    command_throttle: 150ms
    update_interval: 20s

  - modbus_id: modbus_server
    address: 0x01
    server_registers:
      - address: 19
        value_type: S_WORD
        read_lambda: 'return 107;'

      - address: 20
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_errors_bitmask).state);'

      - address: 21
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_capacity_remaining).state);'

      - address: 22
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_total_voltage).state * 100);'

      - address: 23
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_current).state * 100);'

      - address: 24
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_temperature_sensor_1).state * 10);'

      - address: 25
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_max_charge).state * 100) - 500;'

      - address: 26
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_capacity_remaining).state);'

      - address: 27
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_actual_battery_capacity).state);'

      - address: 28
        value_type: S_WORD
        read_lambda: 'return 0;'

      - address: 29
        value_type: S_WORD
        read_lambda: 'return 10;'

      - address: 30
        value_type: S_WORD
        read_lambda: 'return 1;'

      - address: 31
        value_type: S_WORD
        read_lambda: 'return 0;'

      - address: 32
        value_type: S_WORD
        read_lambda: 'return 99;'

      - address: 33
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_bat_count).state * id(jk485_volt_rest).state * 100) + 360;'

      - address: 34
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_errors_bitmask).state);'

      - address: 35
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_max_discharge).state * 100);

```

witoldo7, 2025-06-11:
Hi @test3210-d do you have a Modbus spec for li_4 mode?  Communication work for me, but my inverter GM6200PL ([ANJ620](https://cdn-files.myshopline.com/file/store/1675916196960/ANJ-6200-48PL-WIFI-user-manual.pdf)) do not charge battery when li4 is set

test3210-d, 2025-06-11:
Some of the information was found on GitHub, everything else is reverse engineering. li_4 mode is a mode from the inverter manufacturer, it is not a fact that it is the same for everyone
rs485 jack have 2 line, low speed 1 2 pin 9600 and  7 8 high speed 115200 baud. 
What are the advantages of communication, the exact value of the voltage, the actual value SOC, there are parameters that allow you to charge and discharge by SOC.
pylontech protocol is more universal, but much more difficult to implement for me, there is a pylontech server in esphom, but there is no client for emulation

witoldo7, 2025-06-14:
Above configuration is not correct for SMK Li4 (Lithium Battery Protocol GT Version July 2022 Version 1.0) for example:

SOC != Capacity remaining
```
      - address: 21
        value_type: S_WORD
        read_lambda: 'return int(id(jk485_capacity_remaining).state);'
```

syssi, 2025-07-05:
@test3210-d Could you share your complete YAML configuration? I would like to add a copy to the project.

closed 2025-07-06 by syssi

test3210-d, 2025-07-06:
[yaml.zip](https://github.com/user-attachments/files/21088787/yaml.zip)
Sorry, guys. After the latest bombings of my city by the russians, I haven't really been able time to focus on inverters and batteries