syssi/esphome-smg-ii #124: I can't connect to my SMG II Easun.

TiTTaN40, 2025-09-01:
Hello everyone.
Sorry for my poor English. I'm using a translator.

I can't connect to my SMG II Easun.

Here's my wiring:
[image]

[URL=https://www.casimages.com/i/25090109413917758418631524.jpg.html][IMG]https://nsm09.casimages.com/img/2025/09/01//25090109413917758418631524.jpg[/IMG][/URL]

My Debug Log:

[component:238]: Components should block for at most 30 ms.
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=320 count=8
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=329 count=1
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=331 count=7
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=406 count=1
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=420 count=1
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=643 count=1
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=219 count=2
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=223 count=5
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=229 count=1
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=232 count=3
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=300 count=4
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=305 count=6
[W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=313 count=1
[W][component:237]: Component modbus_controller took a Long time for an operation (60 ms).
[W][component:238]: Components should block for at most 30 ms.

I tried with several ESP32s and TTL converters, and it's the same.

I slightly modified my configuration file:

substitutions: 
name: smg-ii 
device_description: "Monitor and control an ISolar/EASUN SMG II inverter via RS232" 
tx_pin: GPIO16 
rx_pin: GPIO17

esphome: 
name: ${name} 
comment: ${device_description} 
min_version: 2024.6.0 
project: 
name: "syssi.esphome-smg-ii" 
version: 1.3.0

esp32: 
board: esp32dev 
framework: 
type: arduino

web_server: 
port: 80 
version: 1

wifi: 
ssid: !secret wifi_ssid 
password: !secret wifi_password 

manual_ip: 
static_ip: 192.168.1.145 
gateway: 192.168.1.1
subnet: 255.255.255.0
dns1: 192.168.1.1
dns2: 8.8.8.8

I don't understand anything. I've been stuck here for weeks.

Can you help me, please?

TiTTaN40, 2025-09-01:
[image]

Zinyx, 2025-09-03:
Buy this one MAX3232, i have connect to smg-ii-6,2 and work correctly.

[image]

TiTTaN40, 2025-09-03:
I use this one, it should work??

[image]

Zinyx, 2025-09-03:
> I use this one, it should work??
> 
> [image]

For me don't work this one.

TiTTaN40, 2025-09-03:
I've already tried several adapters without converters.

I've already swapped the TX and RX directly on the ESP32 without success.

But I can try again.

syssi, 2025-09-03:
We are talking about different things. Swapping the RX/TX lines on the ESP32 isn't the same as swapping the lines at the RS232 RX/TX side. The MAX3232 has dedicated input (DIN1, RIN1) pins and output pins (DOUT1, ROUT1).

syssi, 2025-09-03:
Please use a continuity tester and verify the connections:

[image]

Please provide a photo of your data cable between the inverter and the rs232 converter too. The connection between the ESP32 and the MAX3232 looks good.

TiTTaN40, 2025-09-03:
I'll test this and get back to you with updates.

TiTTaN40, 2025-09-07:
The TX and RX are OK (pins 11 and 12).

On the schematic, here's the Max 3232's connection to DB9.

[image]

syssi, 2025-09-07:
Could you provide a photo of the data cable between the inverter and max3232 too?

TiTTaN40, 2025-09-07:
I don't understand.
There's no cable between the Max3232 and the inverter.
It's connected directly to it.
Here's a photo of my setup.

[image]

[image]

TiTTaN40, 2025-09-09:
I used the explanations in this video to make my first montage
[https://www.youtube.com/watch?v=cQJcuB6Car8&list=PPSV](url)

TiTTaN40, 2025-09-09:
https://www.youtube.com/watch?v=cQJcuB6Car8&list=PPSV